hku-mars/FAST-LIVO2 #5: When will the source code be available to the public?

Junbug331, 2024-06-24:
Hello, 
I'm excited about FAST-LIVO2 and would love to try FAST-LIVO2 with my devices.

Would you mind informing us when might the source code be available to the public?

Could this be available by this summer?

Thank you.

xuankuzcr, 2024-07-15:
Thank you very much for your interest in our work! I am currently actively addressing the reviewers' comments and revising the paper. Once it is accepted, we will make the source code available (the code is already organized and ready for release).

ZYXiao-g, 2024-09-02:
> Thank you very much for your interest in our work! I am currently actively addressing the reviewers' comments and revising the paper. Once it is accepted, we will make the source code available (the code is already organized and ready for release).

您好，想问下图片中的mesh纹理贴图是怎么做的？后续也会开源这部分工作吗

xuankuzcr, 2024-09-07:
后续会开源的，有两种做法，1是用openmvs的贴图策略，2是用三角面片顶点的颜色插值。