hku-mars/FAST-LIVO2 #6: Hardware used for HKisland01 dataset

gc625, 2024-07-15:
Hi, I was wondering what was the hardware model (camera, gnss, rtk) used to collect data for the HKisland01 aerial dataset?

[image]

xuankuzcr, 2024-07-15:
You can refer to our previous work, "MARS-LVIG dataset: A multi-sensor aerial robots SLAM dataset for LiDAR-visual-inertial-GNSS fusion". This dataset includes a Hikvision CA-050-11UC global-shutter RGB camera equipped with a 5 mm lens. The u-blox ZED-F9P receiver is used to receive raw GNSS measurements, and the RTK receiver is from the DJI M300 RTK UAV.

gc625, 2024-07-16:
Thanks for the quick response! Since there is gps/rtk hardware, is there any integration done with the system to georeference the resulting slam?

xuankuzcr, 2024-07-16:
Currently, our system focuses on LiDAR-Inertial-Visual fusion. The RTK/GPS is used only for ground truth evaluation.

closed 2024-07-16 by gc625