hku-mars/FAST-LIVO2 #7: Hardware Synchronization

pauldeee, 2024-07-15:
Hi there,

Will FAST-LIVO2 require hardware synchronization?

xuankuzcr, 2024-07-16:
FAST-LIVO2 supports both hardware synchronization and non-synchronized modes. However, in the upcoming open-source version, if you are using non-synchronized data, you will need to manually specify the time offset between the LiDAR and the camera in the configuration file (the interface has been provided). We will also release a LIV joint initialization module in this repository later, which will initialize the positions, velocities, biases, and time offsets between LiDAR, camera, and IMU for multiple frames. However, this module is still under development at the moment.

pauldeee, 2024-07-16:
Very cool! Thank you!

closed 2024-07-16 by pauldeee

PanZhichen, 2025-04-01:
> FAST-LIVO2 supports both hardware synchronization and non-synchronized modes. However, in the upcoming open-source version, if you are using non-synchronized data, you will need to manually specify [...]

@xuankuzcr 你好，郑博：
请问这里你提到的“LIV joint initialization module”是指这篇工作吗：《GS-LIVO: Real-Time LiDAR, Inertial, and Visual Multi-sensor Fused Odometry with Gaussian Mapping》

PanZhichen, 2025-04-01:
> FAST-LIVO2 supports both hardware synchronization and non-synchronized modes. However, in the upcoming open-source version, if you are using non-synchronized data, you will need to manually specify [...]

@xuankuzcr 郑博，你好，请问这里提到的 **LIV joint initialization module** 是否有对应的论文？