hku-mars/FAST-LIVO2 #11: About 3DGS

MoonUniverse, 2024-07-29:
Hi，
Thank you for providing such excellent work. Will Fast Livo2 support input data for neural rendering technologies like NeRF and 3DGS? By the way, is there a specific date when we can see the entire complete project?

thanks！

xuankuzcr, 2024-07-29:
Thank you for your attention and support. FAST-LIVO2 supports the generation of input data for neural rendering technologies like NeRF and 3DGS. I have tested it with various methods, including but not limited to f2nerf and 2d-gaussian-splatting. The manuscript will be made public on arXiv after revisions are submitted, and the full code, dataset, and application will be released upon acceptance.

MoonUniverse, 2024-07-29:
Thank you for your reply!

closed 2024-07-29 by MoonUniverse