hku-mars/FAST-LIVO2 #15: 关于下游应用中的无人机飞控算法和路径规划算法

zhangnanyue, 2024-08-30:
感谢贡献这么出色的工作，论文已经拜读完，等待代码的释放。

请教下，关于第十章应用中，无人机的飞控算法，以及路径规划算法Bubble Planner是否有计划开源？

xuankuzcr, 2024-08-31:
感谢你对我们工作的关注！对于第十章无人机全机载自动导航的应用，我会提供所需的软硬件接口，包括UAV上搭载的硬件型号、UAV的CAD模型、onboard版本的轻量级里程计、以及发布imu频率位姿和重力对齐等功能。对于planning和控制，论文里的reference是仅供参考的，具体开源进度可以去对应项目主页查看。但选择哪种控制和planning算法并不是重点，planner你可以使用现在机器人社区里很多用于uav的local planning的算法，比如Fast-Planner。飞控算法可以参考比较常见的MAVROS和PX4-Autopilot。

zhangnanyue, 2024-09-02:
> 感谢你对我们工作的关注！对于第十章无人机全机载自动导航的应用，我会提供所需的软硬件接口，包括UAV上搭载的硬件型号、UAV的CAD模型、onboard版本的轻量级里程计、以及发布imu频率位姿和重力对齐等功能。对于planning和控制，论文里方案是仅供参考的，具体开源进度可以去对应项目主页查看。但选择哪种控制和planning算法并不是重点，planner你可以使用现在机器人社区里用于uav的local planning的算法，比如[ego-planner](https://github.com/ZJU-FAST-Lab/ego-planner)，控制算法可以参考uzh的[MPC](https://github.com/uzh-rpg/rpg_mpc)，飞控算法可以参考比较常见的MAVROS和PX4-Autopilot。

非常感谢您详细的回复，我这边没有其他任何问题了。

祝好！