hku-mars/FAST-LIVO2 #21: Impressive Work! Open Source Plans?

Dumb6, 2024-09-06:
This is really an amazing work. Thank you for your great help to the SLAM community. This work really a pioneer in the field of multi-sensor fusion. When will you consider open-sourcing your code? I can't wait to reproduce your paper results!

ImperiumKontratakuje, 2024-09-06:
I would like to express my deep gratitude for making the FAST-LIVO and FAST-LIVO2 code public.
This is an incredible technology, and I can hardly imagine how much knowledge in mathematics, physics, and computer science is required, as well as how much time must be dedicated to create such an amazing project.
I am also eagerly looking forward to testing FAST-LIVO2 in my robotic constructions.
Every day I start by checking if the publication has been released yet. :)
It’s a bit like waiting for the next launch of the "Starship" with a landing on the moon. :)
Congratulations to the authors of the project. Excellent work!

xuankuzcr, 2024-09-07:
Thank you so much for your kind words and support! We're thrilled by your excitement for FAST-LIVO2. Once the paper is accepted, we’ll make the code open source. We look forward to seeing the results of your tests in your robotic projects. Thanks again for your enthusiasm!