hku-mars/FAST-LIVO2 #23: will LIVO2 optimize the Extrinsics between lidar with camera

jiachenglihxl, 2024-09-15:


xuankuzcr, 2024-09-20:
We will pre-calibrate the pixel-level extrinsic parameters, but we won’t optimize them online as it is prone to negative optimization. In r3live, both the intrinsic and extrinsic parameters are optimized online, with negative optimization being quite evident. I believe an initialization module (e.g., VINS-mono) is quite necessary in the absence of accurate extrinsic parameters.

jiachenglihxl, 2024-09-21:
Thanks! VINS-mono will able to optimize camera extrinsic between IMU. In r3live, I feel we actually get camera or lidar extrinsic between IMU, not the  extrinsic beteen camera with lidar directly. 
However in LIVO system， we can get lidar and camera feature together. So is it possible to use this feature to optimize extrinsic between camera with lidar directly?

xuankuzcr, 2024-10-05:
In LIVO2, the image reference patch is attached to the LiDAR plane. It might be possible to optimize the extrinsic parameters between the camera and LiDAR by minimizing the photometric error of multi-frame observations of the reference patch, leveraging the plane prior.