hku-mars/FAST-LIVO2 #24: 郑博你好 想问下ESIKF同步帧的那部分可以说一些详细的细节吗

1391741823, 2024-09-15:
你好作者 论文中是以IMU为主体帧 想问下筛选雷达帧用了什么方法呢 然后对后面相机帧又用了什么呢

xuankuzcr, 2024-09-20:
1. 将IMU作为主参考系。
2. 在相机采样时刻，将原始超高频的LiDAR点分帧，使LiDAR scan与image frame对齐。
3. 后面没有其他处理了吧，但如果原始数据相机频率很高，比如像hilti一样40Hz，我会隔4个取一帧（一是为了减少计算量和内存，因为prior质量很好，没必要用这么高频率的图像。二是避免分帧后使LiDAR一帧点太少），然后再对LiDAR点分帧。

zhh2005757, 2024-09-27:
> 1. 将IMU作为主参考系。
> 2. 在相机采样时刻，将原始超高频的LiDAR点分帧，使LiDAR scan与image frame对齐。
> 3. 后面没有其他处理了吧，但如果原始数据相机频率很高，比如像hilti一样40Hz，我会隔4个取一帧（一是为了减少计算量和内存，因为prior质量很好，没必要用这么高频率的图像。二是避免分帧后使LiDAR一帧点太少），然后再对LiDAR点分帧。

采用分帧的方式和point lio这种直接计算到点时刻的方式孰优孰劣呢？

xuankuzcr, 2024-10-05:
分帧是为了把相机和LiDAR帧对齐，然后去做sequential update，而point-lio以点频去更新是为了抵抗IMU超量程或饱和的情况，两者目的不同。但是如果只讨论分帧效果的话，在非退化的数据上，分帧确实能减少IMU传播的时间，降低对先验的依赖。