hku-mars/FAST-LIVO2 #28: what is the type of on-board PC on UAV?

nowPB, 2024-10-09:


xuankuzcr, 2024-10-09:
NUC (Intel i7-1360P CPU and 32 GB RAM). The hardware and software setup of this application can be found in Section X-A(1) of the paper.

nowPB, 2024-10-09:
thank you!

> NUC (Intel i7-1360P CPU and 32 GB RAM). The hardware and software setup of this application can be found in Section X-A(1) of the paper.

nowPB, 2024-10-09:
> NUC (Intel i7-1360P CPU and 32 GB RAM). The hardware and software setup of this application can be found in Section X-A(1) of the paper.

What is the power supply voltage and current intensity of this board? Did you design the input power supply yourself or buy it off-the-shelf? I find that the existing mini PC has a large power supply current and voltage, and I can't buy a suitable mobile battery device.

xuankuzcr, 2024-10-11:
1. Rated Voltage and Current: 20V, 6A.
2. Input Power Source: Purchased off-the-shelf.
3. Battery Choice: 6S 5000mAh LiPo battery, with a 20C discharge rate, providing up to 100A—more than enough.
4. Voltage Regulation: Using a buck converter with wide input and 20V output.