hku-mars/FAST-LIVO2 #31: 与GS结合测试

dongxiaozi, 2024-10-12:
您好，想问下，与3DGS结合的测试是直接用的最经典版本的GS吗，还是其他的改进版本呢，我也想用点云直接给到GS进行测试，想取个经，colmap部分好改写吗，或者您改写的GS这部分开源吗，谢谢。

xuankuzcr, 2024-10-13:
做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)

dandelion-chu, 2024-10-14:
> 做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)

您好，请问LIVO2输出colmap格式数据这个功能，能简单说下大概思路么？目前想自己实现这部分，遇到问题了，不知道坐标系是不是有特殊要求？

dandelion-chu, 2024-10-25:
> > > 做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)
> > 
> > 
> > 您好，请问LIVO2输出colmap格式数据这个功能，能简单说下大概思路么？目前想自己实现这部分，遇到问题了，不知道坐标系是不是有特殊要求？
> 
> 您好，请问您实现了吗，目前我实现了通过激光计算的slam位姿转换成colmap格式数据，但是最终3DGS的效果比直接用纯图像做3DGS的还差一点。可以一起交流下。

你好，我这边改了坐标系之后也可以了，我是直接把建图的bag拿过来测试的，一直向前走，训练角度看着还可以吧，可能适合我没有多角度环绕拍摄有关，自由角度没法看

xuezhen2018, 2024-10-28:
> 你好，我这边改了坐标系之后也可以了，我是直接把建图的bag拿过来测试的，一直向前走，训练角度看着还可以吧，可能适合我没有多角度环绕拍摄有关，自由角度没法看

我的训练角度也很好，但这个大概率是过拟合了，PSNR高达35，但是其他角度就一般般，很模糊

huzhanxiaoxue, 2024-10-28:
@xuezhen2018 您好，相关代码可以分享学习学习吗

dandelion-chu, 2024-10-28:
> > 你好，我这边改了坐标系之后也可以了，我是直接把建图的bag拿过来测试的，一直向前走，训练角度看着还可以吧，可能适合我没有多角度环绕拍摄有关，自由角度没法看
> 
> 我的训练角度也很好，但这个大概率是过拟合了，PSNR高达35，但是其他角度就一般般，很模糊

是的，我训练的PSNR也三十多的样子，其他视角很糊
[image]

james201011071, 2024-11-19:
> > > 做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)
> > 
> > 
> > 您好，请问LIVO2输出colmap格式数据这个功能，能简单说下大概思路么？目前想自己实现这部分，遇到问题了，不知道坐标系是不是有特殊要求？
> 
> 您好，请问您实现了吗，目前我实现了通过激光计算的slam位姿转换成colmap格式数据，但是最终3DGS的效果比直接用纯图像做3DGS的还差一点。可以一起交流下。

你好，可以交流私聊一下吗，我也尝试用这种方式做，但是遇到问题了，可以留一下你的微信吗

lihui1998, 2024-11-26:
> > > > 做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)
> > > 
> > > 
> > > 您好，请问LIVO2输出colmap格式数据这个功能，能简单说下大概思路么？目前想自己实现这部分，遇到问题了，不知道坐标系是不是有特殊要求？
> > 
> > 
> > 您好，请问您实现了吗，目前我实现了通过激光计算的slam位姿转换成colmap格式数据，但是最终3DGS的效果比直接用纯图像做3DGS的还差一点。可以一起交流下。
> 
> 你好，可以交流私聊一下吗，我也尝试用这种方式做，但是遇到问题了，可以留一下你的微信吗

换了个git账号，这是我的微信18876643121

lihui1998, 2024-11-26:
> @xuezhen2018 您好，相关代码可以分享学习学习吗

换了个账号，这是我的微信18876643121

closed 2024-12-05 by dongxiaozi

bugmaker1027, 2025-02-14:
> > > 做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)
> > 
> > 
> > 您好，请问LIVO2输出colmap格式数据这个功能，能简单说下大概思路么？目前想自己实现这部分，遇到问题了，不知道坐标系是不是有特殊要求？
> 
> 您好，请问您实现了吗，目前我实现了通过激光计算的slam位姿转换成colmap格式数据，但是最终3DGS的效果比直接用纯图像做3DGS的还差一点。可以一起交流下。

确实要比colmap生成的差很多，我没怎么去优化3dgs，出来以后其他视角渲染结果是这样的

[image]
[image]

上面是fastlivo2输出的结果生成的3dgs渲染结果，下面是colmap重建给的结果，可能是fastlivo2的点太少了，不过空中floater减轻了很多

robotvision2BJTU, 2025-03-07:
> 确实要比colmap生成的差很多，我没怎么去优化3dgs，出来以后其他视角渲染结果是这样的
>
> [image] [image]
>
> 上面是fastlivo2输出的结果生成的3dgs渲染结果，下面是colmap重建给的结果，可能是fastlivo2的点太少了，不过空中floater减轻了很多

你好，麻烦问一下你用哪个3DGS算法测试的，fast-livo2算法导出的colmap结果是txt格式不是bin格式的文件，images.txt没有POINTS2D[] as (X, Y, POINT3D_ID)的信息，在一些方法里还用不起来，需要再把txt处理一下吗

VVivyy, 2025-10-30:
> > > > > 做了一点改进，用法向量对点云的高斯分布进行初始化，使其在平行于物体表面的方向进行优化、分裂和合并。FAST-LIVO2能够直接输出符合Colmap格式的数据，可以直接给到3DGS (这部分开源😉)
> > > > 
> > > > 
> > > > 您好，请问LIVO2输出colmap格式数据这个功能，能简单说下大概思路么？目前想自己实现这部分，遇到问题了，不知道坐标系是不是有特殊要求？
> > > 
> > > 
> > > 您好，请问您实现了吗，目前我实现了通过激光计算的slam位姿转换成colmap格式数据，但是最终3DGS的效果比直接用纯图像做3DGS的还差一点。可以一起交流下。
> > 
> > 
> > 确实要比colmap生成的差很多，我没怎么去优化3dgs，出来以后其他视角渲染结果是这样的
> > [image] ![Image](https://private-user-images.githubusercontent.com/169220442/413192988-7ecae597-5588-4151-af62-4d1e10083121.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjE4MDU3MjIsIm5iZiI6MTc2MTgwNTQyMiwicGF0aCI6Ii8xNjkyMjA0NDIvNDEzMTkyOTg4LTdlY2FlNTk3LTU1ODgtNDE1MS1hZjYyLTRkMWUxMDA4MzEyMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAzMFQwNjIzNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZGQ3ZWZkZjhhZDFkNTg5NWEwMGI1NWZmNzgwNzdhNmQwY2JiZTc5ZTM4MTM0ZWJjNjA4NTE0NzIxMjUyY2Q3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.wHdkWipD6va0TvfXcQViukatv8Oc_EGxL6UGYE30kEs)
> > 上面是fastlivo2输出的结果生成的3dgs渲染结果，下面是colmap重建给的结果，可能是fastlivo2的点太少了，不过空中floater减轻了很多
> 
> 你好，麻烦问一下你用哪个3DGS算法测试的，fast-livo2算法导出的colmap结果是txt格式不是bin格式的文件，images.txt没有POINTS2D[] as (X, Y, POINT3D_ID)的信息，在一些方法里还用不起来，需要再把txt处理一下吗

你好，我用3DGS源码来对fast-livo2导出的数据进行处理，最终在SIBR里看到的相机每一帧的轨迹和真实相差很多，但是重建效果在训练角度很好，怎么解决这个问题呢，相机的轨迹应该是出问题了