hku-mars/FAST-LIVO2 #32: Hardware detail

ita333, 2024-10-12:
Congratulations for the excellent work.
I would like to suggest, if possible, to also focus on the hardware part, so as to allow everyone to create their own handheld 3d scanner, thus sharing the information (3d printing, camblages, material) as they did in a similar project (https://github.com/JanuszBedkowski/mandeye_controller) then using components that are commercially available all over the world and easy to use.
Example:
Lidar (livox mid 360) camera (theta v 360 - gopro) support and battery (directly a gimbal from which to take the power [dji ronin S has 12v outputs on the head and is available at low cost] ) an Rtk sensor for geolocalization would then be perfect to have a geolocalized trajectory and use smartphone or tablet sa monitor [terminal Pc viewer or app].
In the software part the possibility of adding control points to be able to join multiple scans automatically.
Do you have an idea of ​​a possible release date?
congratulations again.

xuankuzcr, 2024-10-13:
We have already [open-sourced our handheld 3D scanner](https://github.com/hku-mars/FAST-LIVO) early on, providing all the technical details, including but not restricted to hardware selection, synchronization scheme, STM32 firmware, and drivers for both the camera and LiDAR. I have personally dedicated significant time and effort to addressing various bugs reported by users, as documented in the [repository’s changelog](https://github.com/sheng00125/LIV_handhold). Additionally, many users have successfully replicated the system and shared their impressive results in issues such as [#32](https://github.com/hku-mars/FAST-LIVO/issues/32), [#96](https://github.com/hku-mars/FAST-LIVO/issues/96), [#110](https://github.com/hku-mars/FAST-LIVO/issues/110), etc. Thank you for your attention.😊

YiChenCityU, 2025-01-08:
Did you support go pro camera?