hku-mars/FAST-LIVO2 #36: 关于点云赋色

wangjiabao123, 2024-10-17:
郑博，
        你好，我想请问一下点云赋色是怎么做的，在Fast-livo里面好像优化位姿后直接使用点云往图像上投影得到RGB，Fast-livo2最后点云赋色有使用patch的信息或者voxelmap里的点信息吗？

xuankuzcr, 2024-10-20:
您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 发布到rviz中进行可视化。相比于后处理的多帧点云上色或使用贝叶斯更新来平滑地图颜色，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。

wangjiabao123, 2024-10-21:
了解，谢谢您的解答！

JinMing-Gu, 2024-10-22:
> 您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 通过优化后的位姿投影到对应图像上进行赋色，然后发布到rviz中进行可视化。相比于**后处理的多帧点云上色**或**使用贝叶斯更新来平滑地图颜色**，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。

请问，“后处理的多帧点云上色或使用贝叶斯更新来平滑地图颜色”分别具体是指什么样的方法，有哪些论文或开源工程使用了这样的方法吗？

ZYXiao-g, 2024-10-22:
> > 您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 通过优化后的位姿投影到对应图像上进行赋色，然后发布到rviz中进行可视化。相比于**后处理的多帧点云上色**或**使用贝叶斯更新来平滑地图颜色**，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。
> 
> 请问，“后处理的多帧点云上色或使用贝叶斯更新来平滑地图颜色”分别具体是指什么样的方法，有哪些论文或开源工程使用了这样的方法吗？

r3live

JinMing-Gu, 2024-10-22:
> > > 您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 通过优化后的位姿投影到对应图像上进行赋色，然后发布到rviz中进行可视化。相比于**后处理的多帧点云上色**或**使用贝叶斯更新来平滑地图颜色**，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。
> > 
> > 
> > 请问，“后处理的多帧点云上色或使用贝叶斯更新来平滑地图颜色”分别具体是指什么样的方法，有哪些论文或开源工程使用了这样的方法吗？
> 
> r3live

好的，感谢解答！

lejj3347, 2025-02-25:
> 您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 通过优化后的位姿投影到对应图像上进行赋色，然后发布到rviz中进行可视化。相比于**后处理的多帧点云上色**或**使用贝叶斯更新来平滑地图颜色**，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。

您好，请问这里赋色后的点就是算法运行时在rviz中看见的地图点嘛？如果是的话，为什么livo2和livo1一样的赋色策略，2的可视化效果会好很多？不是的话，rviz中的地图点是怎么得来的呢。

xuankuzcr, 2025-02-25:
1. 是的 2. 因为livo2的估计出来的相机pose更准

shuttworth, 2025-04-01:
> 您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 通过优化后的位姿投影到对应图像上进行赋色，然后发布到rviz中进行可视化。相比于**后处理的多帧点云上色**或**使用贝叶斯更新来平滑地图颜色**，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。

请问发布到rviz中赋色之后的点云，如何存储到本地呢？

Derkai52, 2025-08-14:
> > 您好，感谢你的问题。fast-livo2的点云上色策略与fast-livo1保持一致，即直接将每帧的原始点 (raw points) 通过优化后的位姿投影到对应图像上进行赋色，然后发布到rviz中进行可视化。相比于**后处理的多帧点云上色**或**使用贝叶斯更新来平滑地图颜色**，这种方式能够更加直观地暴露位姿估计的问题。只要pose存在细微的误差，彩色点云就会出现模糊，从而更快定位问题。
> 
> 请问发布到rviz中赋色之后的点云，如何存储到本地呢？

在对应启动的 xxxx.yaml 文件中，打开：
pcd_save:
  pcd_save_en: true

就能保存在 Log/PCD 下面了，如果没有这个文件夹需要自己创建一下，在程序结束后，一共会自动生成两个点云文件，都是带RGB的点云，只不过一个是降采样的稀疏点云，一个是原始点云。