hku-mars/FAST-LIVO2 #40: 郑博您好，想问一下会有ros2版本么

sanjinna, 2024-10-25:
目前手里有一个nvidia jetson orin nx，不知能否部署的上

xuankuzcr, 2024-10-26:
There isn't a ROS2 version yet, but someone in the community might adapt it in the future. Feel free to submit a pull request. Deployment on Orin NX is supported.

sujit-168, 2025-02-26:
Dr. Zheng, thank you for your excellent work. After two weekends, I migrated a normal and usable ROS2 version on ubuntu22.04 humble. I have used `Retail_Street` and `HKU_Campus` datasets for testing. The results are consistent after migration. I plan to submit a PR to this repository this week. I also adapted the [Vikit](https://github.com/uavfly/vikit.git) module for this migration. 

For well-known reasons, ROS2 does not have a direct global parameter server and a simple method to obtain the corresponding parameters. For details, please refer to https://discourse.ros.org/t/ros2-global-parameter-server-status/10114/11. I use a special way to get camera parameters in `Vikit`. If necessary, I can also submit a PR.