hku-mars/FAST-LIVO2 #43: HILTTI-2022数据集求指教

keleeven, 2024-10-30:
您好，最近在利用fast livo调试HILTTI-2022的数据集，激光里程计已经可以成功运行，但是结合相机这块一直调试有问题，无法正常跟踪特征点，以下是我配置的参数表，麻烦郑博可以点拨一二，感谢！！！
feature_extract_enable : 0
point_filter_num : 1
max_iteration : 10
debug: 1
dense_map_enable : 1
filter_size_surf : 0.05
filter_size_map : 0.1
cube_side_length : 20
grid_size : 40
patch_size : 8
img_enable : 1
lidar_enable : 1
outlier_threshold : 300 # 78 100 156
ncc_en: true
ncc_thre: 100
img_point_cov : 100 # 1000
laser_point_cov : 0.001 # 0.001
delta_time: 0.0

common:
    lid_topic:  "/hesai/pandar"
    imu_topic:  "/alphasense/imu"

preprocess:
    lidar_type: 4 # 1:Livox Avia LiDAR  2:VELO16  3:OUST64  4:XT32 
    scan_line: 32 # 16  64  32
    blind: 1 # blind x m disable

mapping:
    acc_cov_scale: 100
    gyr_cov_scale: 10000
    extrinsic_T: [-0.001, -0.00855, 0.055] # horizon 0.05512, 0.02226, -0.0297
    extrinsic_R: [ 0, -1, 0,
                   -1, 0, 0,
                   0, 0, -1]

pcd_save:
    pcd_save_en: true

camera:
    img_topic: /alphasense/cam0/image_raw
    Rcl: [0,0,-1,
            0,-1,0,
          -1,0,0]
    Pcl: [-0.052, -0.053, 0.068]

xuankuzcr, 2024-10-31:
你用的外参Rcl和Pcl看起来不对，试试这个
Rcl: [ -0.999926, -0.00670802, 0.0101073,
          -0.0100912, -0.00242564, -0.999946,
          0.00673218, -0.999975, 0.00235777]
Pcl: [-0.0549762, 0.0675401, -0.0520599]

keleeven, 2024-10-31:
> 你用的外参Rcl和Pcl看起来不对，试试这个 Rcl: [ -0.999926, -0.00670802, 0.0101073, -0.0100912, -0.00242564, -0.999946, 0.00673218, -0.999975, 0.00235777] Pcl: [-0.0549762, 0.0675401, -0.0520599]

谢谢郑博的回复，试了您的参数配置后，就可以正常跟踪视觉点了，我跑的是exp10_cupola_2.bag这个数据集，但是每次在上楼梯这段狭窄空间的时候，位姿就会发生严重飘逸，并且单独跑LIO也会存在一样的问题，但是看作者的FAST-LIVO2论文实验部分，FAST-LIO2和FAST-LIVO都是很小的姿态误差，所以不太清楚是哪里存在问题，希望郑博可以指点一二，感激！！以下是我FAST-LIVO和FAT-LIO2的参数配置

**FAST-LIVO：**
feature_extract_enable : 0
point_filter_num : 1
max_iteration : 10
debug: 1
dense_map_enable : 0
filter_size_surf : 0.05
filter_size_map : 0.1
cube_side_length : 20
grid_size : 40
patch_size : 8
img_enable : 1
lidar_enable : 1
outlier_threshold : 300 # 78 100 156
ncc_en: false
ncc_thre: 0
img_point_cov : 100 # 1000
laser_point_cov : 0.001 # 0.001
delta_time: -0.05

common:
    lid_topic:  "/hesai/pandar"
    imu_topic:  "/alphasense/imu"

preprocess:
    lidar_type: 4 # 1:Livox Avia LiDAR  2:VELO16  3:OUST64  4:XT32 
    scan_line: 32 # 16  64  32
    blind: 0.5 # blind x m disable

mapping:
    acc_cov_scale: 100
    gyr_cov_scale: 10000
    extrinsic_T: [-0.001, -0.00855, 0.055]
    extrinsic_R: [1.11022302e-16, -1.00000000e+00,  0.00000000e+00,
                  -1.00000000e+00,  1.11022302e-16, -0.00000000e+00,
                  0.00000000e+00,  0.00000000e+00, -1.00000000e+00]

pcd_save:
    pcd_save_en: true

camera:
    img_topic: /alphasense/cam0/image_raw
    Rcl: [ -0.999926, -0.00670802, 0.0101073,
    -0.0100912, -0.00242564, -0.999946,
    0.00673218, -0.999975, 0.00235777]
    Pcl: [-0.0549762, 0.0675401, -0.0520599]
        
**FAST-LIO2：**
common:
    lid_topic:  "/hesai/pandar"
    imu_topic:  "/alphasense/imu"
    time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
    time_offset_lidar_to_imu: 0.0 # Time offset between lidar and IMU calibrated by other algorithms, e.g. LI-Init (can be found in README).
                                  # This param will take effect no matter what time_sync_en is. So if the time offset is not known exactly, please set as 0.0
    
preprocess:
    lidar_type: 3                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line: 32
    timestamp_unit: 3                 # 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 1

mapping:
    acc_cov: 0.1
    gyr_cov: 0.1
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    fov_degree:    360
    det_range:     150.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic
    extrinsic_T: [-0.001, -0.00855, 0.055]
    extrinsic_R: [1.11022302e-16, -1.00000000e+00,  0.00000000e+00,
                  -1.00000000e+00,  1.11022302e-16, -0.00000000e+00,
                  0.00000000e+00,  0.00000000e+00, -1.00000000e+00]

publish:
    path_en:  true
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: false       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

keleeven, 2024-11-01:
尝试了较多的方法，给激光点云建立噪声模型，加上BALM，修改IMU噪声，但无一例外，运行exp10_cupola_2.bag都在上楼梯这段发生了里程计的严重漂移，无论是FAST-LIO2，还是FAST-LIVO，希望郑博抽空可以指点迷津，感激不尽！！！

xuankuzcr, 2024-11-04:
我刚测了下exp10_cupola_2.bag，可以跑下来，顶楼的地面稍微有点厚，但可以接受。livo2里面的LiDAR部分修改自voxel map，参数可以给你参考下。ps: 上下楼梯时很狭窄，voxel_size要设小一点，要不然平面会误判。
```yaml
  max_iterations: 10
  voxel_map_en: true
  pub_plane_en: false
  dept_err: 0.001 
  beam_err: 0.01
  min_eigen_value: 0.005
  sigma_num: 3
  voxel_size: 0.1 
  max_layer: 2
  max_points_num: 150
  layer_init_num: [5, 5, 5, 5, 5]
```
[image]

0ontheroad, 2024-11-04:
> ```yaml
> 10
>   voxel_map_en: true
>   pub_plane_en: false
>   dept_err: 0.001 
>   beam_err: 0.01
> ```

收到！！谢谢郑博抽出时间调试，并感谢细节方面的指导，我去尝试一下。

keleeven, 2024-11-04:
收到！！！谢谢郑博抽出时间调试，并感谢细节上的指点，我去尝试一下。

techiemanvendra, 2025-03-12:
i am using ouster lidar and imu and running by on bag file data but i am able to map the area if i disabled th camera part but if i use camera also then it is not functioning here is my config file please @xuankuzcr @keleeven give some solution
common:
  img_topic: "/camera/top_front/image_raw"
  lid_topic: "/ouster/points"
  imu_topic: "/ouster/imu"
  img_en: 1
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [0.04165, 0.02326, -0.0284]
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  Rcl: [0.001,-0.027,-1.000,
        -1.000,-0.001,-0.001,
        -0.001,1.000,-0.027] 
  Pcl: [-0.021, -0.007, -0.116]

time_offset: 
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess:
  point_filter_num: 1
  filter_size_surf: 0.1
  lidar_type: 3 # Livox Avia LiDAR
  scan_line: 32
  blind: 1

vio:
  max_iterations: 5
  outlier_threshold: 1000 # 78 100 156 #100 200 500 700 infinite
  img_point_cov: 100 # 100 1000 
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: true
  imu_int_frame: 30
  acc_cov: 0.5 # 0.2
  gyr_cov: 0.3 # 0.5
  b_acc_cov: 0.0001 # 0.1
  b_gyr_cov: 0.0001 # 0.1

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.05
  min_eigen_value: 0.0025 # 0.005
  voxel_size: 0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:  
  map_sliding_en: false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: false

publish:
  dense_map_en: true
  pub_effect_point_en: false 
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  seq_name: "CBD_Building_01"
  pose_output_en: false

pcd_save:
  pcd_save_en: true
  colmap_output_en: false # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1
  # how many LiDAR frames saved in each pcd file;
  # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

camera_pinhole.yaml
cam_model: Pinhole
cam_width: 943
cam_height: 367
scale: 0.5
cam_fx: 248.85
cam_fy: 248.85
cam_cx: 471.5
cam_cy: 183.5
cam_d0: 0.0
cam_d1: 0.0
cam_d2: 0.0
cam_d3: 0.0