hku-mars/FAST-LIVO2 #46: 每天过来瞧一眼

lmh12138, 2024-11-29:
苦苦等待开源ing

james201011071, 2024-11-30:
感觉不会开了

xuankuzcr, 2024-12-01:
I've been job hunting recently. I started looking late, and it's already the end of the autumn recruitment season. So please bear with me for a slight delay. I appreciate your understanding.

Gatsby23, 2024-12-02:
> I've been job hunting recently. I started looking late, and it's already the end of the autumn recruitment season. So please bear with me for a slight delay. I appreciate your understanding.

不考虑找教职吗？

1391741823, 2024-12-02:
佬这样都没收到心仪offer吗

defe41251135, 2024-12-02:
佬这样都没收到心仪offer吗

Slam-journey, 2024-12-02:
佬这样都没收到心仪offer吗👀

xincheng-yang, 2024-12-02:
佬这样都没收到心仪offer吗👀

85256638, 2024-12-02:
佬这样都没收到心仪offer吗

Es1erda, 2024-12-03:
佬这样都没收到心仪offer吗:fire:

aditdoshi333, 2024-12-09:
> I've been job hunting recently. I started looking late, and it's already the end of the autumn recruitment season. So please bear with me for a slight delay. I appreciate your understanding.

Hey @xuankuzcr we are hiring for a SLAM Engineer. Would you be interested to jump on a call to discuss further?

85256638, 2024-12-13:
佬找工作怎么样了啊？期待开源

Seroka88, 2025-01-01:
Hey, will this still be open source? Or is it closed source now? PS: Good luck for your job hunting 🔥 @xuankuzcr

rainlord2, 2025-01-15:
非常敬佩大佬的工作，期待分享。

xuankuzcr, 2025-01-16:
The code will be open-sourced before 2025/01/23, I promise!
A big thanks to everyone for your patience and support! 🙏

aditya-inkers, 2025-01-23:
@xuankuzcr Open-sourced before 2025/01/23? 🤔? 

[image]