hku-mars/FAST-LIVO2 #48: 大概什么时间点会开源

Grapgerzhou, 2025-01-16:
郑博，都2025年了，会什么时候开源代码啊？

rainlord2, 2025-01-16:
请耐心一点，大佬时间比较紧张

xuankuzcr, 2025-01-16:
The code will be open-sourced **before 2025/01/23**, I promise!
A big thanks to everyone for your patience and support! 🙏

gloryhry, 2025-01-16:
> The code will be open-sourced **before 2025/01/23**, I promise!
> A big thanks to everyone for your patience and support! 🙏

听到这消息又有盼头了

1391741823, 2025-01-17:
> The code will be open-sourced **before 2025/01/23**, I promise! A big thanks to everyone for your patience and support! 🙏

感谢大佬所做的相关工作，希望大佬工作生活越来越顺利

FengyuGuo, 2025-01-20:
> The code will be open-sourced **before 2025/01/23**, I promise! A big thanks to everyone for your patience and support! 🙏

gift for spring fest!

Zoujingliang, 2025-01-21:
> The code will be open-sourced **before 2025/01/23**, I promise! A big thanks to everyone for your patience and support! 🙏

I will remember this day！！！This is just great! I just finished watching Fast-Livo，hahahahha

beichenlee, 2025-01-21:
感谢大佬的所做的工作，希望大佬工作生活万事如意。今天打开github看到的最好的消息。

IMUTsongX, 2025-01-22:
距离23号0点还有9个小时

Loricon, 2025-01-23:
大佬新年快乐

PreX-AW, 2025-01-23:
> 大佬新年快乐

细节，南方小年

pengjili, 2025-01-23:
距离24日0点还有不到4个小时咯。

PreX-AW, 2025-01-23:
> 距离24日0点还有不到4个小时咯。

永远相信美好的事情即将发生

creativeslamer, 2025-01-23:
睡吧各位，过年多喝点酒

xuankuzcr, 2025-01-23:
Thank you all for your support! 
The code is now open-source 🎉, and the bag files for other data sequences are being uploaded 🚀.

Grapgerzhou, 2025-01-24:
谢谢郑博送给大家的新年礼物。

closed 2025-01-24 by Grapgerzhou