hku-mars/FAST-LIVO2 #50: 关于3D Lidar的选择

a511028896, 2025-01-23:
我想请教一下，在这个算法中，3D Lidar和相机的视场角是不是要接近比较好？因为我想激光点和图像点要尽可能重合比较多，比如说用mid 360等360度的激光雷达，是不是没有Avia这种与相机视场角接近的效果好？

ustc-flicar, 2025-01-24:
我这里有一个其他型号的LiDAR效果参考，我使用的是360度的Velodyne-HDL32激光雷达和一个前视的海康相机，运行Fast-LIVO2的效果也比较不错：

https://github.com/ustc-flicar/ustc-flicar.github.io

[image]

xuankuzcr, 2025-01-29:
是的，一般希望LiDAR和相机的FoV重合得越多越好，这样其实标定也比较容易。Mid360更适合搭配广角鱼眼相机，而Avia则更适合搭配针孔相机。选型没问题的话，两种LiDAR都能得到不错的效果。

123295, 2025-02-05:
郑博你好，请问使用mid360的话用哪一款广角鱼眼相机比较好呢，相比于针孔相机，广角鱼眼相机的时间同步和标定的工作有何差异？

gloryhry, 2025-02-05:
郑博，能否支持多个鱼眼相机与mid360一起工作？修改相机的代码，多加一路相机的线程可以吗？ @xuankuzcr

Yuea922, 2025-02-07:
> 我这里有一个其他型号的LiDAR效果参考，我使用的是360度的Velodyne-HDL32激光雷达和一个前视的海康相机，运行Fast-LIVO2的效果也比较不错：
> 
> https://github.com/ustc-flicar/ustc-flicar.github.io
> 
> [image]

你好，请问相机和雷达的外参是什么，HDL激光和海康相机的标定参数路径中好像没有文件？

xuankuzcr, 2025-02-09:
@123295 可以参考下这个回复 [#121](https://github.com/hku-mars/FAST-LIVO/issues/121)，我还发现了社区里面的关于[fast-livo多相机扩展](https://github.com/cyn4676086/FAST-LIVO-Improvement)，你可以瞅瞅。

ustc-flicar, 2025-02-16:
您好，感谢关注！

USTC FLICAR数据集Velodyne-HDL 32和Hik相机的适配LIVO2参数是这样：

#USTC FLICAR dataset adapted to Fast-LIVO2 external parameters
#Using Velodyne_HDL32 LiDAR and forward-looking Hikision camera

# USTC FLICAR

img_topic: /hik_camera/image_raw
lid_topic: "/velodyne_points_HDL32"
imu_topic: "/imu/data"
  
extrinsic_T: [ 0.0133,0.00245,-0.15381 ] 
extrinsic_R: [0.999696,-0.0242868,-0.00417785,
        0.0242924,0.999704,0.00129262,
        0.00414522,-0.00139371,0.99999]
Rcl:  [0.9986271,  -0.0523854,  -0.0040975,
 -0.0028000,  -0.0248972, -0.9996862,
 0.0523316,  0.9983281,  -0.0250149]
Pcl: [-0.0394, -0.1564, 0.0352]  

cam_model: Pinhole
cam_width: 1440
cam_height: 1080
scale: 0.5
cam_fx: 1841.23617607680
cam_fy: 1840.52373569701
cam_cx: 727.914261021968
cam_cy: 545.179478227874
cam_d0: -0.0423450639246540
cam_d1: 0.0812033992404416
cam_d2: 0.000
cam_d3: -0.000

我从中科大(USTC)本科毕业之后来到港大MaRS读研了，网站后端用的还是中科大的云盘，现在我毕业了没有上传文件更新的权限了，后续我会进行整体迁移和维护，敬请谅解！

关于LIVO2代码适配USTC FLICAR数据集，目前还是以郑博开源出来的代码为准参考，后续我会持续和郑纯然师兄学习交流，并且维护数据集和算法生态，有所更新及时通知您~

https://github.com/ustc-flicar/ustc-flicar.github.io/issues/1

ustc-flicar, 2025-02-16:
> > 我这里有一个其他型号的LiDAR效果参考，我使用的是360度的Velodyne-HDL32激光雷达和一个前视的海康相机，运行Fast-LIVO2的效果也比较不错：
> > https://github.com/ustc-flicar/ustc-flicar.github.io
> > [image]
> 
> 你好，请问相机和雷达的外参是什么，HDL激光和海康相机的标定参数路径中好像没有文件？

您好，感谢关注！

USTC FLICAR数据集Velodyne-HDL 32和Hik相机的适配LIVO2参数是这样：

#USTC FLICAR dataset adapted to Fast-LIVO2 external parameters
#Using Velodyne_HDL32 LiDAR and forward-looking Hikision camera

USTC FLICAR
img_topic: /hik_camera/image_raw
lid_topic: "/velodyne_points_HDL32"
imu_topic: "/imu/data"

extrinsic_T: [ 0.0133,0.00245,-0.15381 ]
extrinsic_R: [0.999696,-0.0242868,-0.00417785,
0.0242924,0.999704,0.00129262,
0.00414522,-0.00139371,0.99999]
Rcl: [0.9986271, -0.0523854, -0.0040975,
-0.0028000, -0.0248972, -0.9996862,
0.0523316, 0.9983281, -0.0250149]
Pcl: [-0.0394, -0.1564, 0.0352]

cam_model: Pinhole
cam_width: 1440
cam_height: 1080
scale: 0.5
cam_fx: 1841.23617607680
cam_fy: 1840.52373569701
cam_cx: 727.914261021968
cam_cy: 545.179478227874
cam_d0: -0.0423450639246540
cam_d1: 0.0812033992404416
cam_d2: 0.000
cam_d3: -0.000

我从中科大(USTC)本科毕业之后来到港大MaRS读研了，网站后端用的还是中科大的云盘，现在我毕业了没有上传文件更新的权限了，后续我会进行整体迁移和维护，敬请谅解！

关于LIVO2代码适配USTC FLICAR数据集，目前还是以郑博开源出来的代码为准参考，后续我会持续和郑纯然师兄学习交流，并且维护数据集和算法生态，有所更新及时通知您~

https://github.com/ustc-flicar/ustc-flicar.github.io/issues/1

Yuea922, 2025-03-01:
> > > 我这里有一个其他型号的LiDAR效果参考，我使用的是360度的Velodyne-HDL32激光雷达和一个前视的海康相机，运行Fast-LIVO2的效果也比较不错：
> > > https://github.com/ustc-flicar/ustc-flicar.github.io
> > > [image]
> > 
> > 
> > 你好，请问相机和雷达的外参是什么，HDL激光和海康相机的标定参数路径中好像没有文件？
> 
> 您好，感谢关注！
> 
> USTC FLICAR数据集Velodyne-HDL 32和Hik相机的适配LIVO2参数是这样：
> 
> #USTC FLICAR dataset adapted to Fast-LIVO2 external parameters #Using Velodyne_HDL32 LiDAR and forward-looking Hikision camera
> 
> USTC FLICAR img_topic: /hik_camera/image_raw lid_topic: "/velodyne_points_HDL32" imu_topic: "/imu/data"
> 
> extrinsic_T: [ 0.0133,0.00245,-0.15381 ] extrinsic_R: [0.999696,-0.0242868,-0.00417785, 0.0242924,0.999704,0.00129262, 0.00414522,-0.00139371,0.99999] Rcl: [0.9986271, -0.0523854, -0.0040975, -0.0028000, -0.0248972, -0.9996862, 0.0523316, 0.9983281, -0.0250149] Pcl: [-0.0394, -0.1564, 0.0352]
> 
> cam_model: Pinhole cam_width: 1440 cam_height: 1080 scale: 0.5 cam_fx: 1841.23617607680 cam_fy: 1840.52373569701 cam_cx: 727.914261021968 cam_cy: 545.179478227874 cam_d0: -0.0423450639246540 cam_d1: 0.0812033992404416 cam_d2: 0.000 cam_d3: -0.000
> 
> 我从中科大(USTC)本科毕业之后来到港大MaRS读研了，网站后端用的还是中科大的云盘，现在我毕业了没有上传文件更新的权限了，后续我会进行整体迁移和维护，敬请谅解！
> 
> 关于LIVO2代码适配USTC FLICAR数据集，目前还是以郑博开源出来的代码为准参考，后续我会持续和郑纯然师兄学习交流，并且维护数据集和算法生态，有所更新及时通知您~
> 
> [ustc-flicar/ustc-flicar.github.io#1](https://github.com/ustc-flicar/ustc-flicar.github.io/issues/1)

好的，非常感谢

DesWudio, 2025-04-28:
> 您好，感谢关注！
>
> USTC FLICAR数据集Velodyne-HDL 32和Hik相机的适配LIVO2参数是这样：
>
> #USTC FLICAR dataset adapted to Fast-LIVO2 external parameters #Using Velodyne_HDL32 LiDAR and forward-looking Hikision camera
>
> USTC [...]

Hello, thanks for your work!

I tried to reproduce your result using the data sequence HF003, however, the reconstruction resut appears less clear than what you had posted.

Could you share the full content in config yaml file?

xl232lx, 2025-08-06:
> [@123295](https://github.com/123295) 可以参考下海康CB系列的M12口的板级相机 + 180度左右的鱼眼镜头。时间同步没啥区别，标定要注意先选好适合大FoV鱼眼的内参模型，要不然外参会一直标不准。

你好，我使用的是mid360+鱼眼相机的组合，内参用Kalibr标定的，外参用DVLC标定一直标不准，请问可以请教一下一般是什么原因引起的么？