hku-mars/FAST-LIVO2 #51: 运行HKairport02.bag报错

keleeven, 2025-01-24:
郑博您好，运行HKairport02.bag一直报错如下
terminate called after throwing an instance of 'std::runtime_error'
  what():  Frame: provided image has not the same size as the camera model
但已经配置好相应的相机参数
cam_model: Pinhole
cam_width: 1224
cam_height: 1024
scale: 0.5
cam_fx: 722.215831395          
cam_fy: 722.171768344 
cam_cx: 588.900539701 
cam_cy: 521.800513284
cam_d0: -0.05729528706141188
cam_d1: 0.1210407244166642
cam_d2: 0.001274128378760289
cam_d3: 0.0004389741530109464

keleeven, 2025-01-24:
把scale改成1也是报一样的错误

x-Yabao, 2025-02-20:
> 郑博您好，运行HKairport02.bag一直报错如下 terminate called after throwing an instance of 'std::runtime_error' what(): Frame: provided image has not the same size as the camera model 但已经配置好相应的相机参数 cam_model: [...]

同学你跑MARS数据集会出现内存使用不断增加然后爆满报错的问题吗

zltan-whu, 2025-02-20:
数据集里面图像是2448x2048，填原始分辨率然后scale设为1，就可以了：

[image]

跑出来效果很nice，我内存不够没有跑完：

[image]

> 郑博您好，运行HKairport02.bag一直报错如下 terminate called after throwing an instance of 'std::runtime_error' what(): Frame: provided image has not the same size as the camera model 但已经配置好相应的相机参数 cam_model: [...]

xuankuzcr, 2025-02-27:
Support for the MARS-LVIG dataset was added in commit 162f843.

1391741823, 2025-03-22:
> 数据集里面图像是2448x2048，填原始分辨率然后scale设为1，就可以了：
>
> [image]
>
> 跑出来效果很nice，我内存不够没有跑完：
>
> [image]

看看显卡有没有用上 可能用的核显 我4g的笔记本gtx1650只用了66% 
安装nvidia prime : `sudo apt install nvidia-prime`   切换独显模式：`sudo prime-select nvidia`