hku-mars/FAST-LIVO2 #56: 安装完成后 roslaunch fast_livo mapping_avia.launch 出现如下段错误，请问是什么原因呢？

Sunshinehaozi, 2025-01-25:
- 非常感谢您开源这项工作，我在运行您代码时遇到了相关问题，希望郑博能帮忙解答

- 环境
- - ubuntu20.04
- 运行指令
- - roslaunch fast_livo mapping_avia.launch
- 报错信息
- coredump堆栈信息

--Type <RET> for more, q to quit, c to continue without paging--c
Core was generated by `/workspace_0/software/ws_fast_livo2/devel/lib/fast_livo/fastlivo_mapping __n'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f2677b8e93b in cv::initUndistortRectifyMap(cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, int, cv::_OutputArray const&, cv::_OutputArray const&) () from /lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2
[Current thread is 1 (Thread 0x7f2662500940 (LWP 281515))]
(gdb) bt

  #0  0x00007f2677b8e93b in cv::initUndistortRectifyMap(cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, int, cv::_OutputArray const&, cv::_OutputArray const&) () at /lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2
#1  0x00007f2677c0e1ac in vk::PinholeCamera::PinholeCamera(double, double, double, double, double, double, double, double, double, double, double, double) () at /workspace_0/software/ws_fast_livo2/devel/lib/libvikit_common.so
#2  0x00005587e884378f in vk::camera_loader::loadFromRosNs(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vk::AbstractCamera*&) ()
#3  0x00007f2679617926 in LIVMapper::initializeComponents() () at /workspace_0/software/ws_fast_livo2/devel/lib/liblaser_mapping.so
#4  0x00007f26796248c8 in LIVMapper::LIVMapper(ros::NodeHandle&) () at /workspace_0/software/ws_fast_livo2/devel/lib/liblaser_mapping.so
#5  0x00005587e8841c09 in main ()

-额外增加的打印（便于看相关数值是否正确）
-  try init PinholeCamera
scale: 0.5
fx_: 646.785
fy_: 646.658
cx_: 313.457
cy_: 261.4
d0: -0.07616
d1: 0.123001
d2: -0.00113
d3: 0.000251
d4: 0
width_: 640
height_: 512

Sunshinehaozi, 2025-01-25:
1. opencv 4.5.5

Idea-in-Dream, 2025-01-26:
may feel opencv version question,I use opencv4.2 is ok

xuankuzcr, 2025-01-29:
感觉楼上说的对，你试试@Sunshinehaozi

bbking2000, 2025-02-14:
> 1. opencv 4.5.5

have you fixed this problem? I have the same question with opencv 4.6.0

cumtAnyu, 2025-02-17:
我也遇到了,请问您解决了嘛?

hyheiyue, 2025-04-04:
蹲解决方案