hku-mars/FAST-LIVO2 #57: Is there any plan to release the part of gaussian splatting?

Allen1droid, 2025-01-26:


xuankuzcr, 2025-01-29:
To enable 3D Gaussian Splatting with FAST-LIVO2’s colored point cloud and poses, simply set:
```bash
pcd_save_en: true
colmap_output_en: true
filter_size_pcd: 0.15 # point cloud downsampling resolution
interval: -1
```
With these settings, the output files will be neatly stored in `./Log/Colmap`. Hope this helps—happy experimenting!

xyj2xx, 2025-02-06:
> To enable 3D Gaussian Splatting with FAST-LIVO2’s colored point cloud and poses, simply set:
> 
> pcd_save_en: true
> colmap_output_en: true
> filter_size_pcd: 0.15 # point cloud downsampling resolution
> interval: -1
> 
> With these settings, the output files will be neatly stored in `./Log/Colmap`. Hope this helps—happy experimenting!
郑博，你好。我在yaml里面改了设置，images.txt会生成相机位姿，但是points3D.txt一直是空的

xuankuzcr, 2025-02-06:
参考下#65，需要建个文件夹