hku-mars/FAST-LIVO2 #58: Request for Hilti Dataset Configuration and Fisheye Lens Support

zzxxtt, 2025-01-26:
Thank you for releasing the code for this impressive LiDAR-inertial-visual odometry system!

I noticed that a lot of the experimental results in the paper are based on the Hilti dataset. I would greatly appreciate it if you could provide the corresponding configuration files used to run FAST-LIVO2 on the Hilti dataset. This would be invaluable for reproducing your results and for further experimentation.

Furthermore, the paper mentions that FAST-LIVO2 supports fisheye lenses. However, after reviewing the code, I haven't found any specific implementations or parameters related to fisheye distortion models. Could you please clarify when the fisheye lens support will be open-sourced, or if this feature is planned for future release? It would be extremely beneficial to leverage this functionality, particularly when using datasets such as the Hilti dataset, which utilizes fisheye cameras.

I am very excited about the potential of FAST-LIVO2 and look forward to your response and future updates to the project.

Thank you for your time and dedication! Wishing you a Happy Chinese New Year!

xuankuzcr, 2025-01-29:
Happy Chinese New Year! Thank you for your interest and support.

The configuration files for the Hilti dataset will be made available in the future. In fact, I have already open-sourced the code implementation for fisheye camera support, which is available at `https://github.com/xuankuzcr/rpg_vikit`. You may not have found it because all camera models share the same function names for projection and back-projection, which are overloaded. 

After the holiday, I will provide example configurations for multi-line LiDAR and fisheye camera setups to facilitate usage. Apologies for the rushed release without proper documentation, and I appreciate your patience!

zzxxtt, 2025-01-31:
Yes I missed that. The fisheye works well after change the camera.yaml like: 
cam_model: EquidistantCamera
.....
k1:
k2:
k3:
k4:

The distortion can directly use the opencv fisheye model.

Again thank you for the great work.

closed 2025-01-31 by zzxxtt

xuankuzcr, 2025-01-31:
You are right! The model can be chosen from `EquidistantCamera`, `OmniCamera`, `PolynomialCamera`, `PinholeCamera`, and `ATANCamera`.