hku-mars/FAST-LIVO2 #60: Poor Performance of HKU Campus Dataset with Original Code and Default Parameters

Lockwuud, 2025-01-27:
Great work! However, I found that I couldn't achieve the same good results as others during the reproduction.
Why is the performance not very good when I directly pull the original code and run it on the HKU campus dataset? The four fonts of the University of Hong Kong are particularly blurry. I haven't modified any parameters.
In fact, I have seen the reproduction results of others, which are clearly better than mine.
My hardware is an x86 platform with a 12th Gen i5 processor and an RTX 3050 GPU.
I have high hopes for the fast-livo2 package and hope you can help me solve this issue.

[image]

ustc-flicar, 2025-01-27:
I guess you may not refer to the calibration.yaml file in the dataset cloud disk to modify the camera-LiDAR external parameters and camera internal parameters corresponding to the HKU Campus Dataset:

# HKU_Campus, CBD_Building_02
Rcl: [-0.00200, -0.99975, -0.02211,
-0.00366, 0.02212, -0.99975,
0.99999, -0.00192, -0.00371]
Pcl: [0.00260, 0.05057, -0.00587]

cam_width: 1280
cam_height: 1024
scale: 0.5
cam_fx: 1176.2874292149932
cam_fy: 1176.21585445307 cam_cx: 592.1187382755453 cam_cy: 509.0864309628322 cam_d0: -0.13218037625958456 cam_d1: 0.15360732717073536 cam_d2: 0.00036918417348059815 cam_d3: -0.00031715324469463964

This is what I ran Fast-LIVO2 with this set of parameters on my laptop (x86 platform with Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, NVIDIA GeForce GTX 1060):  

[image]

[image]

ustc-flicar, 2025-01-27:
As stated in "calibration.yaml": Due to the long development cycle of FAST-LIVO2, we recalibrate the device parameters every three months, so different sequences have different intrinsic and extrinsic parameters.

Therefore, your " I haven't modified any parameters." is wrong operation.

Lockwuud, 2025-01-27:
Thank you! I didn't change the extrinsic and intrinsic parameters. I really appreciate your patient response and guidance.

closed 2025-01-27 by Lockwuud