hku-mars/FAST-LIVO2 #63: Explanation of user-configurable parameters

Josgonmar, 2025-01-29:
Hi @xuankuzcr ,

First of all, thanks for your great work on this project!

I was wondering if you could provide some insights into the user-configurable parameters. While some of them are self-explanatory, others are a bit unclear in terms of their exact effect on performance.

Would it be possible to get a brief explanation of each parameter, especially those that have a significant impact on speed, accuracy, or memory usage? A short guideline on how to tune them for different types of datasets would also be super helpful.

Thanks in advance for your help!

xuankuzcr, 2025-01-29:
Thanks for your kind words and interest!

I originally planned to include detailed explanations for the user-configurable parameters, but due to time constraints, I had to release the project without them. We’re currently on a break for Chinese New Year, but I’ll update the README after the holiday.

Apologies for the rushed release, and thanks for your patience!

Josgonmar, 2025-01-29:
No apologies needed. Enjoy your holidays! 😄