hku-mars/FAST-LIVO2 #65: how can I save the pcd for 3DGS format?

eung-ark, 2025-01-31:
pcd_save:
  pcd_save_en: true
  colmap_output_en: true # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1

I followed your instructions and configured the settings accordingly for my experiment. However, I encountered an error when trying to save the PCD file, as shown in the image below. I suspect that the issue is due to the bag file being too large for my PC to handle the resources properly. Is there any way to resolve this?

Thank you!

+-------------------------------------------------------------+
|                         LIO Mapping Time                    |
+-------------------------------------------------------------+
| Algorithm Stage               | Time (secs)                 |
+-------------------------------------------------------------+
| DownSample                    | 0.002636                    |
| ICP                           | 0.019318                    |
| updateVoxelMap                | 0.002756                    |
+-------------------------------------------------------------+
| Current Total Time            | 0.024891                    |
| Average Total Time            | 0.029424                    |
+-------------------------------------------------------------+
[ INFO] [1738326025.870589303]: Get image, its header time: 1590192781.899781

^C[republish-4] killing on exit
[laserMapping-2] killing on exit
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.terminate called after throwing an instance of 'pcl::IOException'
  what():  : [pcl::PCDWriter::writeBinary] Error during open!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

[image]

Snowbound9173, 2025-01-31:
I think this can be fixed simply by creating the expected directory which is I assume hard coded to "/FAST-LIVO2/Log/PCD" but by cloning the repository you probably will find that this folder isn't pulled.

So just create the PCD folder in that location and try again.

xuankuzcr, 2025-01-31:
You are right! I will fix this issue by creating the `./Log/PCD` folder in advance. Thanks for your discovery!

closed 2025-01-31 by eung-ark