hku-mars/FAST-LIVO2 #66: fout_imu has incomplete type std::ofstream, and laserMapping-2 process has died

hasankiyani007, 2025-01-31:
Errors     << fast_livo:make /home/hasan/catkin_ws/logs/fast_livo/build.make.001.log                                                                                                                      
In file included from /home/hasan/catkin_ws/src/FAST-LIVO2/src/IMU_Processing.cpp:13:
/home/hasan/catkin_ws/src/FAST-LIVO2/include/IMU_Processing.h:52:12: error: field ‘fout_imu’ has incomplete type ‘std::ofstream’ {aka ‘std::basic_ofstream<char>’}
   52 |   ofstream fout_imu;
      |            ^~~~~~~~
In file included from /usr/include/c++/9/ios:38,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /usr/include/eigen3/Eigen/Core:96,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /home/hasan/catkin_ws/src/FAST-LIVO2/include/IMU_Processing.h:16,
                 from /home/hasan/catkin_ws/src/FAST-LIVO2/src/IMU_Processing.cpp:13:
/usr/include/c++/9/iosfwd:119:11: note: declaration of ‘std::ofstream’ {aka ‘class std::basic_ofstream<char>’}
  119 |     class basic_ofstream;
      |           ^~~~~~~~~~~~~~
In file included from /home/hasan/catkin_ws/src/FAST-LIVO2/include/LIVMapper.h:16,
                 from /home/hasan/catkin_ws/src/FAST-LIVO2/src/LIVMapper.cpp:13:
/home/hasan/catkin_ws/src/FAST-LIVO2/include/IMU_Processing.h:52:12: error: field ‘fout_imu’ has incomplete type ‘std::ofstream’ {aka ‘std::basic_ofstream<char>’}
   52 |   ofstream fout_imu;
      |            ^~~~~~~~
In file included from /usr/include/c++/9/ios:38,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /usr/include/eigen3/Eigen/Core:96,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /home/hasan/catkin_ws/src/FAST-LIVO2/include/IMU_Processing.h:16,
                 from /home/hasan/catkin_ws/src/FAST-LIVO2/include/LIVMapper.h:16,
                 from /home/hasan/catkin_ws/src/FAST-LIVO2/src/LIVMapper.cpp:13:
/usr/include/c++/9/iosfwd:119:11: note: declaration of ‘std::ofstream’ {aka ‘class std::basic_ofstream<char>’}
  119 |     class basic_ofstream;
      |           ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/imu_proc.dir/build.make:63: CMakeFiles/imu_proc.dir/src/IMU_Processing.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2650: CMakeFiles/imu_proc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/laser_mapping.dir/build.make:63: CMakeFiles/laser_mapping.dir/src/LIVMapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:274: CMakeFiles/laser_mapping.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/hasan/catkin_ws/build/fast_livo; catkin build --get-env fast_livo | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

..........................................................................................................................................................................................................
Failed     << fast_livo:make                        [ Exited with code 2 ]                                                                                                                                
Failed    <<< fast_livo                             [ 11.2 seconds ]           

First, for anyone facing the former error of fout_imu has incomplete type, add:
#include <fstream> 
to the top of the Imu_processing.h file.
Secondly, Im getting the laser mapping is died error when running roslaunch fast_livo mapping_avia.launch. Rviz doesnt shut down, but playing the bagfile doesnt show anything in Rviz. Anything I've done wrong? Sorry if it's a beginner mistake, I'm new to this. :)

xuankuzcr, 2025-02-01:
Check all cv::Mat initializations in your code and ensure rows and cols are positive. Add debug prints before each initialization:
```cpp
std::cout << "Mat size: " << rows << " x " << cols << std::endl;
if (rows <= 0 || cols <= 0) {
    std::cerr << "Error: Invalid matrix size!" << std::endl;
}
cv::Mat img(rows, cols, CV_8UC3);

```

hasankiyani007, 2025-02-01:
I haven't changed anything in the source code really, and I'm not providing my own data, neither have I allocated any memory for the image. I am attempting to build and test FAST-LIVO2 on the datasets provided, namely CFD_Building_01 and Retail_Street. Why this issue then?

hasankiyani007, 2025-02-01:
Solved the issue, problem was multiple opencv installations conflicting with each other. Noticed the warning when rebuilding my workspace. 
I removed all opencv installations, then reinstalled opencv 4.2, then built and tested fast livo, its working.

closed 2025-02-01 by hasankiyani007

Cureser, 2025-02-25:
> Solved the issue, problem was multiple opencv installations conflicting with each other. Noticed the warning when rebuilding my workspace. I removed all opencv installations, then reinstalled opencv 4.2, then built and tested fast livo, its working.

My situation is similar to yours, can I see if you can see how you look when you run it with opencv4.2, especially HKU_Campus dataset

crazy-robotics, 2025-02-26:
@hasankiyani007  I have `opencv 4.2.0` installed i get the same error #110