hku-mars/FAST-LIVO2 #67: 100% CPU consumption

crazy-robotics, 2025-02-03:
I am testing `Fast-LIVO2` on my local PC `13th Gen Intel® Core™ i5-13420H × 12 ` with `8GB` RAM. I attached a screenshot below when checking on `htop` all cores are at `100%` usage and checking `top` my `cpu` is at `480%` usage. How can I implement `Fast-LIVO2` on an indoor UAV having  `jetson orin nano 8GB` or `orange pi 5 plus` with `32GB` RAM, Is that sufficient? It is computationally expensive how can I overcome this problem 

[image]

xuankuzcr, 2025-02-03:
1. **FAST-LIVO2 uses a maximum of 4 threads.** In your screenshot, **only two threads** are being used by FAST-LIVO2.
2. The 480% CPU usage comes from the rviz visualization, as clearly indicated in your screenshot.
3. Both the Orin NX and RK3588 are more than enough.
4. There’s **no need** to overcome any problems (luckily, we’ve done the hard work for you).

Thanks for your interest.

crazy-robotics, 2025-02-03:
Thank you @xuankuzcr