hku-mars/FAST-LIVO2 #68: Excess RAM Usage

hasankiyani007, 2025-02-04:
I have 64 GB RAM installed. On the HIT Graffiti Wall Sequence, which is of 8 minutes, out of 64 gb, around 40-45 GB RAM is consumed during this mapping process. Is there any way to optimize RAM Usage, or to offload the currently built map onto the storage drive, so that we can free up RAM and continue mapping for an extended amount of time, if we were to be providing data in real time to the framework, instead of just recording a rosbag?
I see you guys used an onboard pc on the drone in one of your videos, what model was it? How did you manage the mapping there with limited RAM and GPU power?

[image]

Furthermore, another issue, which may be slightly irrelevant in regards to FAST-LIVO2, is that on a pc with integrated graphics graphics but enough RAM, the Rviz Visualization window complete freezes up. To my knowledge, the Rviz panels are GPU-intensive tasks, which display and load the map in the viewer. Is there any generic way to optimize this or to make this viewer more efficient? 

And great work by the way! Thanks so much for making this open source.

xuankuzcr, 2025-02-04:
I used the default parameters and all the raw points, and it only consumed **3.7GB** of memory. I hope the "40-45 GB RAM" you mentioned is not a deliberate exaggeration. Thanks for your interest and attention.
[image]

hasankiyani007, 2025-02-04:
[image]

Not at all a deliberate exaggeration sir. This is the RAM Usage as you can see in the picture above at this point, at around 250 seconds into the grafitti wall sequence. At this point, its consumed 25 gb of ram, but stopped there due to a random segmentation fault and the process died. My last run which completed the entire bag file of 487 seconds, consumed around 40 GBs of RAM. I haven't changed any settings at all.

hasankiyani007, 2025-02-04:
[image]

Here is another screenshot with the complete sequence completed. The mapping process consumed 40.1 GBs, and Rviz consumed 7 GBs. The sequence is the 20 Gb HIT_Graffiti_Wall_01.bag sequence.

hasankiyani007, 2025-02-04:
I have a Livox Avia Sensor, I tested FAST-LIO on that for real-time mapping. It consumed very less memory, for around a minute of mapping an indoor hall, it consumed only 300-400 Mb. I'll test it on the bagfiles and also an extensive area for FAST-LIO and will update when done. 
Still, the memory usage is extremely high for FAST-LIVO2.

xuankuzcr, 2025-02-04:
Everything you said sounds so unbelievable. I'm sorry that I couldn't help you.

hasankiyani007, 2025-02-04:
Kindly reopen the issue, as maybe someone else can help. Thank you

xuankuzcr, 2025-02-05:
@hasankiyani007, please continue in this issue. Pay attention to your attitude and avoid spamming. Your previous topic contained misleading information—I am absolutely sure the issue is due to your own configuration. Instead of being persistent in blaming others, take the time to check your own setup first. Also, please respect the work of others. Nothing is owed to you.

Your other questions have already been discussed in previous issues. You might find the answers you need in #63 and #65 if you take a moment to look.

We appreciate your cooperation and efforts in maintaining a healthy community.

hasankiyani007, 2025-02-05:
Hello kind sir. I do apologize. I genuinely did not blame you or your work. My only intent when posting was to try to find a fix for my problem. Actually, I was assuming the RAM Usage was similar to previous frameworks, and more importantly, I did think the error may be on my side when you first replied. I have much to learn, I am sorry for my mistakes and will try to be better, which is the most I can do. I will check those issues as well, I may have missed them. I once again apologize, and thank you for your patience.

hasankiyani007, 2025-02-05:
To provide a bit of context for everyone else, I have 2 devices that I have set up for this. The first is my personal pc, with a Ryzen 5 5600 processor, 16 GB RAM, and a dedicated Nvidia GTX 1660S graphic card. The second is an Asus NUC 13 pro kit, with an intel i9 1340p processor, with 64 GB RAM, and integrated graphics.
The software environments on both are Ubuntu 20.04, Ros Noetic, PCL 1.10, Eigen version 3.3.7, OpenCV version 4.2.0. I am testing FAST-LIVO2 on these devices. I followed the exact git read me process to build fast-livo2, with some slight changes as needed, which I'll quickly describe on both:
I installed Ros Noetic, PCL, Eigen, and opencv on my devices. Then, I added Sophus, made a small change as mentioned [here](https://github.com/uzh-rpg/rpg_svo/issues/237#issuecomment-426409963)  and installed it. I also added the rpg_vikit to my caktin workspace. I used catkin build instead of catkin_make to build my workspace. However, on both devices I'm facing similar issues. I've tested fast-livo1 and r3live on my personal pc as well, and it also consumed extensive amounts of RAM for large sequences, to the point where my pc would freeze. Similar issues were reported by other users for R3LIVE and FAST-LIVO1 as well, and the general consensus there was the map was being saved in the memory and held there, leading to such high utilization.
However, since everyone else is getting no usage issues, what may I have done wrong, or can anyone help me get some debugging information so I can test my systems? Maybe I'm using the wrong versions of any software, or any other issue? Thank you all in advance.

dongjq23, 2025-02-05:
> 为了为其他所有人提供一些上下文，我为此设置了2个设备。第一个是我的个人PC，配备Ryzen 5 5600处理器，16 GB RAM和专用的NVIDIA GTX 1660S图形卡。第二个是Asus nuc 13 Pro套件，带有Intel I9 1340p处理器，具有64 GB RAM，并集成图形. 两者上的软件环境都是Ubuntu 20.04，Ros Noetic，PCL 1.10，Eigen版本3.3.7，OpenCV版本4.2.0。我正在这些设备上测试Fast-Livo2。我遵循了确切的git读我的过程来构建快速livo2，并根据需要进行一些微小的更改，我会在这两个方面迅速描述: 我在设备上安装了Ros Noetic，PCL，Eigen和OpenCV。然后，我添加了Sophus，如前所述进行了少量更改 [这里](https://github.com/uzh-rpg/rpg_svo/issues/237#issuecomment-426409963) 并安装了它。我还将RPG_VIKIT添加到我的Caktin工作区。我使用Catkin Build而不是Catkin_make来构建工作空间。但是，在这两个设备上，我都面临类似的问题。我也在个人PC上也测试了Fast-Livo1和R3Live，并且还消耗了大量的大型RAM，以至于我的PC会冻结。其他用户也向R3Live和Fast-Livo1报告了类似的问题，并且总体共识保存在内存中并保留在那里，导致了如此高的利用率. 但是，由于其他所有人都没有遇到使用问题，我可能会做错什么，或者有人可以帮助我获得一些调试信息，以便我可以测试我的系统？也许我正在使用任何软件的错误版本或其他任何问题？提前谢谢大家.

您好，我的电脑配置（显卡是RTX4050）和软件库版本可以在下面图片中看到，设置point_filter_num = 3时，程序运行占用内存为3.8G，结果供您参考。

[image]
[image]
[image]

hasankiyani007, 2025-02-05:
Just a quick update so that the thread does not get spammed:
@pcl5  Thank you for your help. Since you mentioned you built without mimalloc, I went ahead and manually removed the mimalloc installation, and rebuilt my catkin workspace, and now the RAM usage is normal, that is, the complete sequence used 4-5 GB. I have yet to test further. I will continue testing this on other devices with the same solution, and will update and close the issue with this solution within the next 2 days if the excess usage is not reproduced. Thank you for the information @dongjq23 as well, much appreciated everyone who is helping!

eung-ark, 2025-02-05:
I had the same symptoms to be honest. But as you mentioned above, I removed all the mimalloc and rebuilt it too, and then RAM level came back to normal. Maybe I built mimalloc wrong or something.. :D
Thanks to the help of the community.

G4419, 2025-02-05:
On my 127-second dataset, fastlivo_mapping consumed about 20GB of RAM. After uninstalling mimalloc, the RAM usage returned to normal. Therefore, this issue appears to be related to mimalloc.

xuankuzcr, 2025-02-05:
FAST-LIVO2, unlike R3LIVE and its predecessor FAST-LIVO, represents a qualitative leap in memory efficiency, computation time, and robustness, thanks to its sparse map structure and optimized system design, as detailed in the paper. Therefore, it should not be generalized indiscriminately (as you have seen its performance).

The high performance of mimalloc comes at the cost of additional memory caching, which is likely causing the issue. However, the impact of memory consumption is not obvious on my computer, and the reason for this is still unclear. For now, it is recommended not to install this library.

Additionally, if the current memory usage is still unsatisfactory, you can set `map_sliding_en = true` (default settings use a global map). This will keep memory usage fluctuating within a narrow range instead of continuously increasing.

Thank you all for your support and validation! 🙌🎉

hasankiyani007, 2025-02-07:
**Final Update:**
 I tested FAST-LIVO2 on another system, with and without mimalloc. With mimalloc, I had the same 40 GB Ram Usage on the 8 minute sequence, and after removing mimalloc and testing FAST-LIVO2 without it, the same 8 minute sequence had around 7 GB of usage. Please note, this usage is excluding the RAM Usage by RViz of course, which may tend to also add up as well. I will consider this issue solved. Thank you everyone for the help! 

**RAM USAGE WITHOUT MIMALLOC:**

[image]

closed 2025-02-07 by hasankiyani007