hku-mars/FAST-LIVO2 #70: using FAST-LIVO2 with only VIO mode

hdrive, 2025-02-06:
在 FAST-LIVO1 中，img_en: 1, lidar_en: 0, 可以进入 VIO 模式，但是在 FAST-LIVO2 中，SLAM_MODE 不包含 VIO 模式。我是想单独运行 VIO 看看跑的效果如何。谢谢！

xuankuzcr, 2025-02-06:
您好，感谢您的关注和支持！`SLAM_MODE`里目前只包含`LO, LIO, LIVO`三种模式，而FAST-LIVO1中也不是严格的`VIO`模式，因为FAST-LIVO2中的visual map points的空间位置是由LiDAR map points提供的，因此LiDAR部分必须存在。

zhipengak, 2025-02-06:
是否可以在LIO更新部分state不加solution，以及不更新state的cov是否能达到只有（visual map points的空间位置是由LiDAR map points提供的）VIO在起作用，这样方便单独验证VIO是否正常

xuankuzcr, 2025-02-07:
@zhipengak 你可以尝试一下，但这样的验证方法不太合理：

fast-livo2中的vio并不是对视觉slam里面方法的生搬照抄。为了提升效率，fast-livo2中的视觉部分设计并使用的是**一步到位**的frame-to-map稀疏直接法，而不是svo中的frame-to-frame稀疏直接法和frame-to-map光流法相结合。可以这么设计的原因是，fast-livo2融合得很紧，充分利用LiDAR部分提供的较好先验分布 (state prior)，避免视觉部分的光度误差优化陷入局部最优 (即论文中的ESIKF with sequential update)。因此直接禁用LiDAR部分的状态更新，会破坏这个系统的第一性原理。

除此之外，在我们的数据中，图像频率仅为10Hz (而svo通常是30Hz)，图像间的视差很大。如果禁用LiDAR部分的状态更新，会加剧刚才提到的"一步到位frame-to-map"的问题。

如果想验证VIO的效果，在退化场景下，比较LIO与LIVO两种模式的表现会更为合理。(其他相关讨论见这个issue [#170](https://github.com/hku-mars/FAST-LIVO/issues/170)）

zhipengak, 2025-02-07:
非常感谢你的回复！你开源的代码对整个SLAM行业有巨大的推进作用！

zhipengak, 2025-02-07:
非常感谢你的回复！
看了你的回复我明白了很多，而且经过验证在跑数据包时候的确在部分LIO退化的场景下，LIVO还能够维持精准的定位。
但是我还是比较疑问的是：
如果在激光退化较长时间情况下（例如下面图片的大片空地上），假如激光只有20米量程，而前面有40米区域都是开阔这种情况，这时候设备往前走的过程中视觉的frame-to-map是否会因为imu的误差和图像稀疏直接法的误差导致map的地图点出现较大的误差，最后传递到frame-to-map时候图像的误差，然后逐渐误差累加，在激光没有打到对面的有效轮廓之前就飞掉这种情况。
当然我测试过fast LIVO在LiDAR_Degenerate这个数据包跑，激光长时间退化情况下定位是没有出现问题的，因为视觉这块维持住了定位，但是图像10Hz的情况下稀疏直接法是否能足够维持住定位，这个应该是很有挑战性的。是否有可能在我这张图片展示的户外场景下出现我上面分析的这种情况？你觉得会不会？
我正在搭建设备来复现这个算法，非常感谢你的分享！！
[image]

xuankuzcr, 2025-02-08:
如果只做frame-to-map的稀疏直接法，可能会出现分层的现象，但应该也和旋转的角速度有关。你说的这种情况也不太合理，至少近处20m内的地面会有LiDAR点，不会一点约束都没有，视觉测量只要约束住退化方向就可以。我觉得是能handle住的，具体以数据为准。

hdrive, 2025-02-08:
好的，谢谢郑博的解答，非常感谢！

closed 2025-02-08 by hdrive