hku-mars/FAST-LIVO2 #73: MID360作为硬件能跑FAST-LIVO2吗

ChunqiuYang, 2025-02-06:
您好，非常感谢您开源这么优秀的工作，从Fast-lio, r2live,r3live...一直关注你们实验室的工作到现在，有非常多非常优秀的项目，想问一下您，有测试过MID360作为Lidar源对FAST-LIVO2算法的适配性吗，或者MID360跑FAST-LIVO2算法的话有什么需要注意的点呢，非常感谢！

Idea-in-Dream, 2025-02-12:
您好，你用mid360跑通了嘛，效果是什么样的呢，我的效果一般，可以交流一下嘛

fjz0911fang123, 2025-02-12:
如果多扫一会应该会更好，我也用mid 360 

发自我的iPhone

ChunqiuYang, 2025-02-18:
> 您好，你用mid360跑通了嘛，效果是什么样的呢，我的效果一般，可以交流一下嘛
您好，我还没开始测试，主要还是无人机用，所以要考虑小型化，目前正准备测试mid360，有做时间硬同步吗

ChunqiuYang, 2025-02-18:
> 如果多扫一会应该会更好，我也用mid 360&nbsp;
> 
> 发自我的iPhone
> […](#)

定位效果好吗

fjz0911fang123, 2025-02-19:
> > 如果多扫一会应该会更好，我也用mid 360 
> > 发自我的iPhone
> > […](#)
> 
> 定位效果好吗

我绕小区跑了一圈没啥问题，比fastlio2 要好很多（fastlio2最后Z轴飘了10m多）， 关于硬件时间同步，是参考https://blog.csdn.net/qq_40301351/article/details/142202297和https://github.com/hku-mars/Livox_handheld  ，你的相机得支持外部信号触发

[image]

ChunqiuYang, 2025-02-19:
> [@ChunqiuYang](https://github.com/ChunqiuYang) 感谢支持，测过很多Mid360数据，我跑效果挺好的啊 [hku-mars/FAST-LIVO#96](https://github.com/hku-mars/FAST-LIVO/issues/96) 跑得有问题的话可以把rosbag发给我一下，我来定位下是数据问题还是算法问题。

好的，非常感谢郑博您的回复，目前我正准备组装硬件来测试，到时有相关的问题我及时向您反馈~

ChunqiuYang, 2025-02-19:
> 我绕小区跑了一圈没啥问题，比fastlio2 要好很多（fastlio2最后Z轴飘了10m多）， [...]

太赞了！我今天组装好硬件测试一下，迫不及待了

20211105, 2025-02-21:
@fjz0911fang123 你好，请问你用的什么品牌相机呀

YanhaoZhang, 2025-03-28:
> 我绕小区跑了一圈没啥问题，比fastlio2 要好很多（fastlio2最后Z轴飘了10m多）， [...]

您好 这个链接打不开了 请问可以再分享下吗