hku-mars/FAST-LIVO2 #76: How to filter out the effects of dynamic objects

Lockwuud, 2025-02-07:
This is a really great project! I used the lidar and imu of a livox mid360 as input to run fast-livo2 and got great builds, but it seems to be affected by dynamic objects, in other words, it incorporates moving objects into the builds, for example, all the trajectories of a moving person in the field are built, here are some examples, is there a way to get rid of the dynamic objects please? I am very much looking forward to your reply!

[image]

xuankuzcr, 2025-02-09:
Thanks for your support! Maybe you can filter dynamic point clouds by using semantic information and object velocity.

Lockwuud, 2025-02-12:
Compared to map building, I value the odometry performance more. After conducting some simple tests, it seems that while dynamic objects are mapped, they do not significantly affect the accuracy of the odometry. I would like to seek your advice: in order to achieve better odometry output, should I adjust the parameters of the voxel filter or other parameters? I look forward to your response.