hku-mars/FAST-LIVO2 #77: Occasional IMU callback stalls and data loss causing large timestamps gaps

G4419, 2025-02-07:
During the execution of the IMU callback function, there is an occasional issue where the program stalls briefly, causing data loss. As a result, the **timestamp** from the callback can significantly exceed the previous **last_timestamp_imu**, leading to abnormal behavior. 

[image]

xuankuzcr, 2025-02-09:
The issue occurs due to high playback speeds of the bag file or insufficient hard drive interface bandwidth. It has been fixed now.

G4419, 2025-02-11:
> I have encountered this issue as well, and for me, it seems to be mainly related to the parameters `patch_size` and `patch_pyrimid_level`. Inappropriate parameters may freeze the program.

I wasn't using VIO mode when the issue occurred, so it likely isn't the problem you're referring to. As for Dr. Zheng's newly submitted version, I haven't tested it yet.

closed 2025-02-14 by G4419