hku-mars/FAST-LIVO2 #81: 2D coordinates of detected features

stas-polukeev, 2025-02-10:
First of all, thanks for the great work, results are amazing!
I wanted to ask a question:
In the exported colmap files in images.txt the points2d[] are hard cooded to be 0.0 0.0 -1
I wanted to add this line, so that all detected points are imported into the file. In the rviz interface the points are shown on the image, so there is a place in the code, where the frame and detected points are crossed, so why can't we export them into the colmap file?

I tried looking into it, and found out that new_frame_ in vio.cpp has list of Features, that should contain the needed Vector2d px_ which I wanted to use for export, however
when i tried printing out each frame's features i found out that the list is empty, so the features are not stored?

i added to vio.cpp in dumpDataForColmap() 
```
std::cout << new_frame_->nObs();
for (Feature* feature : new_frame_->fts_)
   std::cout << feature->px_ << std::endl;
```
the nObs shows zero, so the printing in the for loop is not even happening. 
I am a bit confused, and sorry if the question is trivial, cpp is not my main language, so some things might not be as obvious for me.

To wrap it up:

1. Is it possible to export 2d coordinates of the detected points for each frame into colmap's images.txt?
2. If it is possible, why is frame features are empty?
3. should i look somewhere else in the code?

Thanks in advance

xuankuzcr, 2025-02-11:
The map points exist directly in `feat_map` rather than in a `frame` because our map is surfel-based, not frame-based. You can export 2D pixel coordinates using the points in `feat_map`, but keep in mind that these points are sparse. For a denser result, modify `publish_frame_world` to process `dumpDataForColmap()`.

JiaoxianDu, 2025-02-11:
I have encountered this issue as well, and for me, it seems to be mainly related to the parameters `patch_size` and `patch_pyrimid_level`. Inappropriate parameters may freeze the program.

stas-polukeev, 2025-02-11:
@xuankuzcr Thanks for a swift reply!
I still have a few questions:
1. Is it possible to just use the PlotTrackedPoints() function in vio.cpp? as it has both the image and the 2d coordinates
2. I see that feat_map has visual points, each of them has an instance of Feature class.  I see px_ is what i need and but i don't see where can i find image id for matching? should i look into patch information?
3. will this still work considering the downsampling in the end? Will it still be possible to understand which image sees which centroid from the downsampled map? how exactly can this be done, could you give any guidance?

once again, sorry if the questions are trivial, the project is really complex as it's out of my usual scope, however it's really amazing