hku-mars/FAST-LIVO2 #83: Reconstruct mesh from R3live

eung-ark, 2025-02-11:
Hello, I'm surprised by the tremendous test results every day. Thank you.

I have a question.

I'm sorry if this is a novice question.
Can I apply r3live_reconstruct_mesh from R3live? I want to make it with textured mesh.

Thank you always.

xuankuzcr, 2025-02-11:
r3live uses Delaunay triangulation, but the mesh's edges and corners tend to stick together. It is unsuitable for our dense and accurate point cloud. I have provided an example for TSDF-based mesh construction, and you can use the following code.
```python
import os
import glob
import natsort
import vdbfusion
import numpy as np
import open3d as o3d
from plyfile import PlyData

class Dataset:
    def __init__(self, folder: str):
        super().__init__()
        self.scan_files = natsort.natsorted(glob.glob(os.path.join(folder, "*.ply"))) 
        self.poses = np.array([np.eye(4) for _ in range(len(self.scan_files))]) 

    def __getitem__(self, idx):
        if idx >= len(self.scan_files):
            raise IndexError("Index out of range")
        
        pose = np.linalg.inv(self.poses[0]) @ self.poses[idx]
        points, _ = self.read_ply(self.scan_files[idx])
        points = np.array(points, dtype=np.float64)
        return points, pose

    def __len__(self):
        return len(self.scan_files)
    
    def read_ply(self, ply_file):
        ply_data = PlyData.read(ply_file)
        vertex = ply_data['vertex']
        points = np.c_[vertex['x'], vertex['y'], vertex['z']]
        colors = np.c_[vertex['red'], vertex['green'], vertex['blue']]
        return points, colors

if __name__ == '__main__':
    source_dir = "/home/chunran/Downloads/Dataset/"
    
    vdb_volume = vdbfusion.VDBVolume(voxel_size=0.02, sdf_trunc=0.1)

    dataset = Dataset(source_dir)

    for i in range(len(dataset)):
        scan, origin = dataset[i]
        vdb_volume.integrate(scan, origin)

    vert, tri = vdb_volume.extract_triangle_mesh(min_weight=0.1)

    mesh = o3d.geometry.TriangleMesh(
        o3d.utility.Vector3dVector(vert),
        o3d.utility.Vector3iVector(tri),
    )

    mesh.compute_vertex_normals()
    o3d.visualization.draw_geometries([mesh])

    o3d.io.write_triangle_mesh("output_mesh.ply", mesh)
```

shingkim, 2025-02-13:
Hi, how to get these .ply files?

xuankuzcr, 2025-02-15:
@shingkim Please check the updated code for textured mesh reconstruction.