hku-mars/FAST-LIVO2 #87: 启用map_sliding_en之后,里程计会发生漂移

848451399, 2025-02-13:
1. 这个是没开启

2.这是开启之后 sliding_thresh=8 half_map_size=200

3..这是开启之后 sliding_thresh=100 half_map_size=200

为什么体素地图多次滑动更新之后里程计会发生漂移

IMUTsongX, 2025-02-14:
你好郑博，我同样遇到这个问题，和他的效果是一样的，以下是没有开启之前的效果，定位很精准。

[image]

[image]
开启之后飞机飞回来会有很大的偏差，以下是测试效果；

[image]

[image]

xuankuzcr, 2025-02-14:
参数需要微调下，要不然历史地图的root voxel删除太多，容易产生drift。