hku-mars/FAST-LIVO2 #88: 关于不同类型雷达适配

goldqiu, 2025-02-13:
请问下FAST-LIVO2支持机械旋转式LiDAR吗，我的理解是这种雷达的FOV一般是30°x360°，在点数相同的情况下，与相机重合部分上的点数会比固态式LiDAR少很多，这会影响该算法的估计精度以及3D场景建模的效果吧。还有就是算法本身应该没有提供处理机械旋转式LiDAR的接口吧。

Es1erda, 2025-02-13:
FAST-LIVO2支持机械旋转式激光雷达，一般这类FOV的雷达会配置鱼眼相机使用以增加共视FOV。 按照我的理解，共视FOV小会导致相机视场外的点被浪费，但不太会影响算法的odometry和上色效果。

xuankuzcr, 2025-02-14:
#50

ustc-flicar, 2025-02-16:
> 请问下FAST-LIVO2支持机械旋转式LiDAR吗，我的理解是这种雷达的FOV一般是30°x360°，在点数相同的情况下，与相机重合部分上的点数会比固态式LiDAR少很多，这会影响该算法的POSE估计精度以及3D场景建模的效果吧。 谢谢！

您好！我使用的是360度的Velodyne-HDL32激光雷达和一个前视的海康相机，运行Fast-LIVO2的效果也比较不错：

https://github.com/ustc-flicar/ustc-flicar.github.io

[image]

我们在郑博上面附带的#50 issue里有更多的讨论~

goldqiu, 2025-02-17:
> FAST-LIVO2支持机械旋转式激光雷达，一般这类FOV的雷达会配置鱼眼相机使用以增加共视FOV。 按照我的理解，共视FOV小会导致相机视场外的点被浪费，但不太会影响算法的odometry和上色效果。

感谢您的回答！

goldqiu, 2025-02-17:
> 您好！我使用的是360度的Velodyne-HDL32激光雷达和一个前视的海康相机，运行Fast-LIVO2的效果也比较不错：
>
> https://github.com/ustc-flicar/ustc-flicar.github.io
>
> [image]
>
> 我们在郑博上面附带的[#50](https://github.com/hku-mars/FAST-LIVO2/issues/50) [...]

感谢您的回答！我们要在乘用车上复现该算法，后续有问题加入讨论！

2380675303, 2025-02-18:
请问如果使用mid360及其他360度的机械激光，如果想达到更好的算法效果，有哪些广角或者鱼眼相机推荐吗