hku-mars/FAST-LIVO2 #89: 计算激光点的测量误差的协方差

fjz0911fang123, 2025-02-13:
郑博您好！
      在看VoxelMapManager::BuildVoxelMap()函数中计算激光点测量误差的协方差的代码，发现其与《Efficient and Probabilistic Adaptive Voxel Mapping for Accurate Online LiDAR Odometry》论文中公式（3），有些许不一致，郑博有时间帮忙看一下

[image]

fjz0911fang123, 2025-02-20:
看了代码中好几个地方有计算激光点误差协方差（激光本体坐标系的看明白了，就是转到世界系下看不懂），都跟《Efficient and Probabilistic Adaptive Voxel Mapping for Accurate Online LiDAR Odometry》论文中的计算公式不一致，不知道是不是我理解的有问题，望郑博帮忙解惑，感谢！

xuankuzcr, 2025-02-20:
没事，具体哪部分代码啊

fjz0911fang123, 2025-02-20:
郑博，这个两个地方
[image]
[image]

[image]

fjz0911fang123, 2025-02-24:
郑博，有空帮忙看一下

fjz0911fang123, 2025-03-07:
好的，谢谢郑博！

Lmh-sys, 2025-03-25:
> 好的，谢谢郑博！

您好，我正在学习FAST-LIVO2，可以分享一下你的代码注释吗？我不太看得懂，一直在钻研，感激不尽！lmhyx660@outlook.com

gyyc233, 2025-05-28:
我也遇到这个疑问，看了VoxelMap也是这么写的，可以参考 https://github.com/hku-mars/VoxelMap/issues/15

zhengshunkai, 2026-02-05:
> 好的，谢谢郑博！

请问一下，你这边的改好了吗？求参考一下