hku-mars/FAST-LIVO2 #91: Odometry Instability Wide Streets & Corridors – Large Oscillations

BenderBot-es, 2025-02-13:
First, thank you for this amazing work! I’m using FAST-LIO2 with a rotating multi-plane LiDAR, and it performs very well. Great job! 
I recently opened a small [PR](https://github.com/hku-mars/FAST-LIVO2/pull/90) to fix an issue I encountered during testing.

However, I’m facing odometry instability (sometimes it oscillates back and forth along the X-axis very abruptly and over large distances, tens of meters) when running in wide streets with parallel buildings and  corridors (tunnel effect). I suspect this is due to point cloud degeneration. Do you have any recommended configurations to improve stability in these environments?
Also, for large-scale outdoor/indoor scenarios, what settings would you recommend for a rotating LiDAR + IMU?

Thanks.

xuankuzcr, 2025-02-14:
Thanks for your PR! 
For degenerate cases, add visual constraints.
For outdoor, increase `voxel_size` and relax `min_eigen_value`.
For indoor, reduce `voxel_size` and tighten `min_eigen_value`.
Additionally, `min_eigen_value` can be progressively relaxed as the octree level increases.

BenderBot-es, 2025-02-14:
Thank you very much for your config recommendations, I will try them these days.

The problem with using visual info is that all the 360º lidar points that does not overlap with the camera FoV are discarded and the performance worsens compared to using only lidar, but all the points. Would there be a way to use visual info in the overlapping points and only lidar in the rest?

xuankuzcr, 2025-02-15:
Haha, actually, in the current version, LiDAR points outside the camera's FoV are still utilized and not discarded. They just aren't displayed in the Rviz visualization. Let me know if you have further questions!

BenderBot-es, 2025-02-15:
I'm glad to know that non-overlapping points are not discarded!

Then, I must be doing something wrong because when I add the camera, the odometry worsens significantly and vibrates a lot. My extrinsic calibration is probably not good. I'll review it—Is there a recommended method for calibration?

I've also tried the parameter modifications you suggested, but the odometry still drifts by tens of meters in corridors and tunnel areas. I hope to solve it with well-calibrated visual data!

Thank you very much!