hku-mars/FAST-LIVO2 #93: Output point number: 0

Shaoyule, 2025-02-14:
hi 郑博！
我在gazebo中搭建了一个仿真环境，录制了bag包，跑fastlivo2总会在17s左右崩溃，

[image]

我尝试单独跑lio部分  
img_en: 0
lidar_en: 1
仍然在17s左右崩溃

[image]

我怀疑是雷达问题，于是我跑了fastlio2，可以完整跑下来

[image]

期待回复！

xuankuzcr, 2025-02-14:
Hi，麻烦把bag发一下~

Shaoyule, 2025-02-14:
从QQ邮箱发来的超大附件

camera_pinhole_shaoyu.yaml (391B, 无限期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=326530364c98cc9b06126c731366014b19115102070556054c5c530e004b0553055c1d54575f034951005104500302540250060335783307000855445439430d0f0d5f5a5039400c000a4943154e064d4f1c515b59660e&code=ae065f3d

iris2.bag (1.18G, 2025年05月15日 17:25 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=0f363762f2c7fb8156416b2716300a1a4942515303540f561c010154021d0f0c00021a0156060c180305075108060b000805025a3039385c435f44501e525952310b&code=167b0085

Shaoyule, 2025-02-14:
> Hi，麻烦把bag发一下~

已发送

xuankuzcr, 2025-02-14:
@Shaoyule 没收到哇，请发我邮箱或者po在issue里，记得提供LiDAR和IMU的外参

xuankuzcr, 2025-02-15:
问题已修复，具体见commit 7951a2f

[image]

Shaoyule, 2025-02-16:
> 问题已修复，具体见commit [7951a2f](https://github.com/hku-mars/FAST-LIVO2/commit/7951a2f69ac9431c27669fe87ef9944378524c65)
> 
> [image]

嗯嗯，我也能成功跑了，谢谢郑博！

closed 2025-02-20 by Shaoyule