hku-mars/FAST-LIVO2 #94: Position Estimation from FAST-LIVO2 to px4

crazy-robotics, 2025-02-14:
I am using pixhawk 4 with [px4](https://github.com/PX4/PX4-Autopilot) verison 1.15.2 , onboard computer jetson nano and [Livox mid 360](https://www.livoxtech.com/mid-360) Lidar trying to use [FAST-LIVO2](https://github.com/hku-mars/FAST-LIVO2). But I dont know how to fuse external position from [Livox mid 360](https://www.livoxtech.com/mid-360) to Pixhawark. I read following [discussion](https://discuss.px4.io/t/external-position-with-px4/12205) on `external position with PX4`  they say change the parameter `EKF2_AID_MASK` from `0` to `24` for external postion fuse to `EKF2` but I dont' find `EKF2_AID_MASK` in my parameter list I share my parameter list below
```
# Onboard parameters for Vehicle 1
#
# Stack: PX4 Pro
# Vehicle: Unknown
# Version: 1.15.2 
# Git Revision: 4817c0618a000000
#
# Vehicle-Id Component-Id Name Value Type
ASPD_SCALE_1	1.000000000000000000	9
ATT_EN	0	6
BAT1_A_PER_V	36.367515563964843750	9
BAT1_CAPACITY	-1.000000000000000000	9
BAT1_I_CHANNEL	-1	6
BAT1_N_CELLS	0	6
BAT1_R_INTERNAL	0.004999999888241291	9
BAT1_SOURCE	0	6
BAT1_V_CHANNEL	-1	6
BAT1_V_CHARGED	4.050000190734863281	9
BAT1_V_DIV	18.100000381469726563	9
BAT1_V_EMPTY	3.599999904632568359	9
BAT1_V_LOAD_DROP	0.100000001490116119	9
BAT2_A_PER_V	36.367515563964843750	9
BAT2_CAPACITY	-1.000000000000000000	9
BAT2_I_CHANNEL	-1	6
BAT2_N_CELLS	0	6
BAT2_R_INTERNAL	0.004999999888241291	9
BAT2_SOURCE	-1	6
BAT2_V_CHANNEL	-1	6
BAT2_V_CHARGED	4.050000190734863281	9
BAT2_V_DIV	18.100000381469726563	9
BAT2_V_EMPTY	3.599999904632568359	9
BAT2_V_LOAD_DROP	0.100000001490116119	9
BATMON_DRIVER_EN	0	6
BAT_AVRG_CURRENT	15.000000000000000000	9
BAT_CRIT_THR	0.070000000298023224	9
BAT_EMERGEN_THR	0.050000000745058060	9
BAT_LOW_THR	0.150000005960464478	9
BAT_N_CELLS	3	6
BAT_V_CHARGED	4.050000190734863281	9
BAT_V_EMPTY	3.599999904632568359	9
BAT_V_LOAD_DROP	0.300000011920928955	9
BAT_V_OFFS_CURR	0.000000000000000000	9
CAL_ACC0_ID	3670026	6
CAL_ACC0_PRIO	50	6
CAL_ACC0_ROT	-1	6
CAL_ACC0_XOFF	0.135726585984230042	9
CAL_ACC0_XSCALE	1.000000000000000000	9
CAL_ACC0_YOFF	0.084648445248603821	9
CAL_ACC0_YSCALE	1.000000000000000000	9
CAL_ACC0_ZOFF	-0.074879199266433716	9
CAL_ACC0_ZSCALE	1.000000000000000000	9
CAL_ACC1_ID	3932170	6
CAL_ACC1_PRIO	50	6
CAL_ACC1_ROT	-1	6
CAL_ACC1_XOFF	0.137242138385772705	9
CAL_ACC1_XSCALE	1.000000000000000000	9
CAL_ACC1_YOFF	0.057041298598051071	9
CAL_ACC1_YSCALE	1.000000000000000000	9
CAL_ACC1_ZOFF	-0.305777311325073242	9
CAL_ACC1_ZSCALE	1.000000000000000000	9
CAL_ACC2_ID	0	6
CAL_ACC3_ID	0	6
CAL_AIR_CMODEL	0	6
CAL_AIR_TUBED_MM	1.500000000000000000	9
CAL_AIR_TUBELEN	0.200000002980232239	9
CAL_BARO0_ID	0	6
CAL_BARO1_ID	0	6
CAL_BARO2_ID	0	6
[... 785 lines cut ...]
UAVCAN_SV_FUNC6	0	6
UAVCAN_SV_FUNC7	0	6
UAVCAN_SV_FUNC8	0	6
UAVCAN_SV_MAX1	1000	6
UAVCAN_SV_MAX2	1000	6
UAVCAN_SV_MAX3	1000	6
UAVCAN_SV_MAX4	1000	6
UAVCAN_SV_MAX5	1000	6
UAVCAN_SV_MAX6	1000	6
UAVCAN_SV_MAX7	1000	6
UAVCAN_SV_MAX8	1000	6
UAVCAN_SV_MIN1	0	6
UAVCAN_SV_MIN2	0	6
UAVCAN_SV_MIN3	0	6
UAVCAN_SV_MIN4	0	6
UAVCAN_SV_MIN5	0	6
UAVCAN_SV_MIN6	0	6
UAVCAN_SV_MIN7	0	6
UAVCAN_SV_MIN8	0	6
UAVCAN_SV_REV	0	6
USB_MAV_MODE	2	6
UXRCE_DDS_CFG	0	6
VT_B_DEC_MSS	2.000000000000000000	9
WV_EN	
```

xuankuzcr, 2025-02-15:
Perhaps you should upgrade the PX4 firmware: use the latest PX4 version (1.15.3 or newer) to access `EKF2_AID_MASK`.

crazy-robotics, 2025-02-18:
> Perhaps you should upgrade the PX4 firmware: use the latest PX4 version (1.15.3 or newer) to access `EKF2_AID_MASK`.

@xuankuzcr I have following parameter [EKF2_EV_CTRL](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_EV_CTRL) , [EKF2_HGT_REF](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_HGT_REF) , [EKF2_EV_DELAY](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_EV_DELAY) , [EKF2_EV_POS_X](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_EV_POS_X) , [EKF2_EV_POS_Y](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_EV_POS_Y), [EKF2_EV_POS_Z](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_EV_POS_Z) in `PX4 V 1.15.2` which is mention in following doc [Vision or Motion Capture Systems for Position Estimation](https://docs.px4.io/main/en/ros/external_position_estimation.html#ekf2-tuning-configuration) but how can I pass the `Position estimate` from `FAST-LIVO2` to `PX4`

xuankuzcr, 2025-02-18:
Using `/mavros/vision_pose/pose` provided by `FAST-LIVO2` is sufficient. MAVROS will automatically convert the data from this topic into a `VISION_POSITION_ESTIMATE` MAVLink message and send it to PX4.