hku-mars/FAST-LIVO2 #95: fatal error when build the rpg_vikit and Fast-LIVO2

Victorrr1, 2025-02-15:
First of all, thanks to Dr.Zheng for such high-quality work.

Recently, I met a tricky problem when I tried to reproduce the work. Tons of errors are thrown during the catkin_make process, such as:
`
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:192:3: error: reference to ‘Matrix’ is ambiguous
  192 |   Matrix<double, DIM_STATE, 1> operator-(const StatesGroup &b)
      |   ^~~~~~
In file included from /opt/ros/melodic/include/sophus/rotation_matrix.hpp:7,
                 from /opt/ros/melodic/include/sophus/so3.hpp:4,
                 from /opt/ros/melodic/include/sophus/se3.hpp:4,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:21,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/visual_point.h:17,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/src/visual_point.cpp:13:
/opt/ros/melodic/include/sophus/types.hpp:37:7: note: candidates are: ‘template<class Scalar, int M, int N> using Matrix = Eigen::Matrix<Scalar, M, N>’
   37 | using Matrix = Eigen::Matrix<Scalar, M, N>;
      |       ^~~~~~
In file included from /usr/include/eigen3/Eigen/Core:252,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/utils/so3_math.h:4,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:16,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/visual_point.h:17,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/src/visual_point.cpp:13:
/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:70:9: note:                 ‘template<class _Scalar, int _Rows, int _Cols, int _Options, int _MaxRows, int _MaxCols> class Eigen::Matrix’
   70 | > class Matrix;
      |         ^~~~~~
In file included from /home/nuc/catkin_ws/src/FAST-LIVO2/include/visual_point.h:17,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/src/visual_point.cpp:13:
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:220:3: error: reference to ‘Matrix’ is ambiguous
  220 |   Matrix<double, DIM_STATE, DIM_STATE> cov; // states covariance
      |   ^~~~~~
In file included from /opt/ros/melodic/include/sophus/rotation_matrix.hpp:7,
                 from /opt/ros/melodic/include/sophus/so3.hpp:4,
                 from /opt/ros/melodic/include/sophus/se3.hpp:4,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:21,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/visual_point.h:17,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/src/visual_point.cpp:13:
/opt/ros/melodic/include/sophus/types.hpp:37:7: note: candidates are: ‘template<class Scalar, int M, int N> using Matrix = Eigen::Matrix<Scalar, M, N>’
   37 | using Matrix = Eigen::Matrix<Scalar, M, N>;
      |       ^~~~~~
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:136:11: error: ‘struct StatesGroup’ has no member named ‘cov’
  136 |     this->cov(6, 6) = 0.00001;
      |           ^~~
In file included from /usr/include/eigen3/Eigen/Core:252,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/utils/so3_math.h:4,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:16,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/include/visual_point.h:17,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/src/visual_point.cpp:13:
/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:70:9: note:                 ‘template<class _Scalar, int _Rows, int _Cols, int _Options, int _MaxRows, int _MaxCols> class Eigen::Matrix’
   70 | > class Matrix;
      |         ^~~~~~
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:137:11: error: ‘struct StatesGroup’ has no member named ‘cov’
  137 |     this->cov.block<9, 9>(10, 10) = MD(9, 9)::Identity() * 0.00001;
      |           ^~~
In file included from /home/nuc/catkin_ws/src/FAST-LIVO2/include/visual_point.h:17,
                 from /home/nuc/catkin_ws/src/FAST-LIVO2/src/visual_point.cpp:13:
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h: In constructor ‘StatesGroup::StatesGroup()’:
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:135:11: error: ‘struct StatesGroup’ has no member named ‘cov’
  135 |     this->cov = MD(DIM_STATE, DIM_STATE)::Identity() * INIT_COV;
      |           ^~~
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:137:60: error: lvalue required as left operand of assignment
  137 |     this->cov.block<9, 9>(10, 10) = MD(9, 9)::Identity() * 0.00001;
      |                                                            ^~~~~~~
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h: In copy constructor ‘StatesGroup::StatesGroup(const StatesGroup&)’:
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:149:11: error: ‘struct StatesGroup’ has no member named ‘cov’
  149 |     this->cov = b.cov;
      |           ^~~
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:149:19: error: ‘const struct StatesGroup’ has no member named ‘cov’
  149 |     this->cov = b.cov;
      |                   ^~~
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h: In member function ‘StatesGroup& StatesGroup::operator=(const StatesGroup&)’:
/home/nuc/catkin_ws/src/FAST-LIVO2/include/common_lib.h:161:11: error: ‘struct StatesGroup’ has no member named ‘cov’
[... 284 lines cut ...]
make[2]: *** Waiting for unfinished jobs....
FAST-LIVO2/CMakeFiles/vio.dir/build.make:103: recipe for target 'FAST-LIVO2/CMakeFiles/vio.dir/src/visual_point.cpp.o' failed
make[2]: *** [FAST-LIVO2/CMakeFiles/vio.dir/src/visual_point.cpp.o] Error 1
FAST-LIVO2/CMakeFiles/vio.dir/build.make:75: recipe for target 'FAST-LIVO2/CMakeFiles/vio.dir/src/vio.cpp.o' failed
make[2]: *** [FAST-LIVO2/CMakeFiles/vio.dir/src/vio.cpp.o] Error 1
CMakeFiles/Makefile2:7087: recipe for target 'FAST-LIVO2/CMakeFiles/vio.dir/all' failed
make[1]: *** [FAST-LIVO2/CMakeFiles/vio.dir/all] Error 2
FAST-LIVO2/CMakeFiles/lio.dir/build.make:75: recipe for target 'FAST-LIVO2/CMakeFiles/lio.dir/src/voxel_map.cpp.o' failed
make[2]: *** [FAST-LIVO2/CMakeFiles/lio.dir/src/voxel_map.cpp.o] Error 1
CMakeFiles/Makefile2:7113: recipe for target 'FAST-LIVO2/CMakeFiles/lio.dir/all' failed
make[1]: *** [FAST-LIVO2/CMakeFiles/lio.dir/all] Error 2
[ 43%] Linking CXX shared library /home/nuc/catkin_ws/devel/lib/libnumpy_eigen_test.so
Copying library files to python directory
'/home/nuc/catkin_ws/devel/lib/libnumpy_eigen_test.so' -> '/home/nuc/catkin_ws/devel/lib/python2.7/dist-packages/numpy_eigen/libnumpy_eigen_test.so'
[ 43%] Built target numpy_eigen_test
[ 43%] Linking CXX shared library /home/nuc/catkin_ws/devel/lib/libnumpy_eigen.so
Copying library files to python directory
'/home/nuc/catkin_ws/devel/lib/libnumpy_eigen.so' -> '/home/nuc/catkin_ws/devel/lib/python2.7/dist-packages/numpy_eigen/libnumpy_eigen.so'
[ 43%] Built target numpy_eigen
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

`
I am pretty sure that I followed the installation steps, the only difference I can remember is the error indicating "#include<Sophus/se3.h> no such file" and I changed all the se3.h with se3.hpp. Is the error above related to the wrong version of sophus or something?

xuankuzcr, 2025-02-15:
Hi, you didn't follow the README instructions correctly. You should use the Sophus non-templated/double-only version. Make sure to run `git checkout a621ff`.

Victorrr1, 2025-02-15:
> Hi, it looks like you didn’t follow the README installation instructions correctly. Please use the non-templated/double-only version of Sophus and make sure to run `git checkout a621ff`.

Yes, you're right. Besides, when using the non-templated Sophus,  CMakeLists of vikit_common need to add "set(Sophus_LIBRARIES libSophus.so)". Thanks for your help. I'll close this issue soon.

closed 2025-02-15 by Victorrr1

crazy-robotics, 2025-02-26:
I came across following error while compiling 
``` 
[ 85%] Built target test_vk_common_patch_score
[ 90%] Linking CXX executable /home/mubashir/fast_livo2/devel/lib/vikit_common/test_vk_common_camera
[ 95%] Linking CXX executable /home/mubashir/fast_livo2/devel/lib/vikit_common/test_vk_common_triangulation
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator=(Sophus::SE3 const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3()'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Sophus::SE3 const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Sophus::SE3 const&) const'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SO3::matrix() const'
collect2: error: ld returned 1 exit status
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/test_vk_common_camera.dir/build.make:167: /home/mubashir/fast_livo2/devel/lib/vikit_common/test_vk_common_camera] Error 1
make[1]: *** [CMakeFiles/Makefile2:501: rpg_vikit/vikit_common/CMakeFiles/test_vk_common_camera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator=(Sophus::SE3 const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3()'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Sophus::SE3 const&)'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Sophus::SE3 const&) const'
/usr/bin/ld: /home/mubashir/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SO3::matrix() const'
collect2: error: ld returned 1 exit status
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/build.make:167: /home/mubashir/fast_livo2/devel/lib/vikit_common/test_vk_common_triangulation] Error 1
make[1]: *** [CMakeFiles/Makefile2:933: rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/all] Error 2
[100%] Linking CXX shared library /home/mubashir/fast_livo2/devel/lib/libvikit_ros.so
[100%] Built target vikit_ros
make: *** [Makefile:141: all] Error 2
Invoking "make -j12 -l12" failed

```
and solve it by adding `set(Sophus_LIBRARIES libSophus.so)` in `CMakeLists.txt` file in `vikit_commo`