hku-mars/FAST-LIVO2 #98: tunnel数据有计划放出来吗？

Dorothy-2016, 2025-02-19:
Hi 郑博：
           论文中的隧道数据有打算开源吗？我自己录了一个隧道数据，跑出来的轨迹变形了，且出现了一定的退化情况：

[image]

[image]

这里附上了数据的链接，麻烦郑博有空的时候帮忙看一下，感谢！
链接: [https://pan.baidu.com/s/1TYdPU8jfAc7fwsA5uxFL8Q?pwd=p923](url)提取码: p923 
--来自百度网盘超级会员v4的分享

另，一起附上了一个大桥的数据，也是出现了退化的现象，针对此类特征点少的场景，郑博有什么改善的调试建议吗？

Dorothy-2016, 2025-02-20:
是的！刚发现上传上去的有点问题：

[image]

原始标定出来的外参如下：
extrinsic_T: [-0.003642, -0.105453, -0.004306]
extrinsic_R: [0.999226, -0.001323, 0.039306, 0.000996, 0.999965, 0.008338, -0.039316, -0.008293, 0.999192]

根据上次long-term的调试建议我将R手动改成了单位阵测试，上传云盘时删减一些注释掉的东西把这个给不小心改了==，已更新(手动抱拳！)
前述问题中出现的轨迹变形和退化现象是在正确外参下跑出的结果，另，我这个数据采集设备上配的是左右摄像头，所以单个摄像头与激光的共视肯定少一半，这样是不是会损失掉部分激光的有效特征点而影响跟踪稳定性，反而不如只开激光来定位？

[image]

[image]

如图，隧道这组数据，加上视觉后当前cloud_registered的点云明显少很多，注册的map也少一半。不过这组数据无论是否开启视觉，最后都出现了一定程度的退化

[image]

xuankuzcr, 2025-02-20:
你重新测一下吧，我就直接用你给的参数文件跑的 (关掉`pcd_save_en`)，效果相当惊艳

xuankuzcr, 2025-02-20:
你这个`tunnel.bag`持续怼墙退化，怼了21min，看起来像一个海底隧道

xuankuzcr, 2025-02-20:
21min:38s的超长隧道

[image]
[image]
[image]

26min:12s的超长大桥

[image]

Dorothy-2016, 2025-02-20:
好嘞，我再试一下，方便把您跑出来的轨迹txt发我一份嘛，我对比看一下

20211105, 2025-02-20:
@Dorothy-2016 请问您用的相机是哪一款呀？

Dorothy-2016, 2025-02-21:
> [@Dorothy-2016](https://github.com/Dorothy-2016) 请问您用的相机是哪一款呀？

您好，是自己组装的

20211105, 2025-02-21:
@Dorothy-2016 能方便透漏一下这个相机产品型号吗

Dorothy-2016, 2025-02-21:
> [@Dorothy-2016](https://github.com/Dorothy-2016) 能方便透漏一下这个相机产品型号吗

您好，是自己组装的样机，没有型号

RockyYHL, 2025-05-22:
> Hi 郑博： 论文中的隧道数据有打算开源吗？我自己录了一个隧道数据，跑出来的轨迹变形了，且出现了一定的退化情况：
> 
> [image]
> 
> [image]
> 
> 这里附上了数据的链接，麻烦郑博有空的时候帮忙看一下，感谢！ 链接: [https://pan.baidu.com/s/1TYdPU8jfAc7fwsA5uxFL8Q?pwd=p923](url)提取码: p923 --来自百度网盘超级会员v4的分享
> 
> 另，一起附上了一个大桥的数据，也是出现了退化的现象，针对此类特征点少的场景，郑博有什么改善的调试建议吗？

您好，这份长隧道数据可以再分享下吗，感激不尽！