hku-mars/FAST-LIVO2 #101: 跑MARS_LVIG数据集内存泄露

x-Yabao, 2025-02-20:
郑博您好：
我在跑MARS_LVIG数据集时，内存用量会逐渐变大爆满，最后报错
```
[laserMapping-2] process has died [pid 14454, exit code -9, cmd /home/nvidia/open_source/fast_livo2_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/home/nvidia/.ros/log/f0fcbcea-ef25-11ef-a2fe-b46dc2cb9176/laserMapping-2.log].
log file: /home/nvidia/.ros/log/f0fcbcea-ef25-11ef-a2fe-b46dc2cb9176/laserMapping-2*.log
```
用的是ORIN NX，内存16GB，请问是哪里出现问题了呢

x-Yabao, 2025-02-20:
配置文件如下：
```
common:
  img_topic: "/left_camera/image"
  lid_topic: "/livox/lidar"
  imu_topic: "/livox/imu"
  img_en: 1
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [0.04165, 0.02326, -0.0284]
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  # MARS_LVIG HKisland HKairport
  Rcl: [0.00438814,-0.999807,-0.0191582,
        -0.00978695,0.0191145,-0.999769,
        0.999942,0.00457463,-0.00970118]
  Pcl: [0.016069, 0.0871753, -0.0718021]
  # MARS_LVIG AMtown AMvalley
  # Rcl: [0.00298068,-0.999735,-0.0231428,
  #      -0.00504595,0.0231320,-0.999740,
  #       0.999985,0.00309701,-0.00497598]
  # Pcl: [-0.0025563, 0.0567484, -0.0512149] 

time_offset: 
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess:
  point_filter_num: 2 #1
  filter_size_surf: 0.15
  lidar_type: 1 # Livox Avia LiDAR
  scan_line: 6
  blind: 5  #0.8

vio:
  max_iterations: 5
  outlier_threshold: 1000 # 78 100 156 #100 200 500 700 infinite
  img_point_cov: 100 # 100 1000 
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: true
  imu_int_frame: 30
  acc_cov: 0.5 # 0.2
  gyr_cov: 0.3 # 0.5
  b_acc_cov: 0.0001 # 0.1
  b_gyr_cov: 0.0001 # 0.1

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.05
  min_eigen_value: 0.0025 # 0.005
  voxel_size: 0.6 #0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:  
  map_sliding_en: false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: false

publish:
  dense_map_en: false #true
  pub_effect_point_en: false 
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  seq_name: "CBD_Building_01"
  pose_output_en: false

pcd_save:
  pcd_save_en: false
  colmap_output_en: false # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1
  # how many LiDAR frames saved in each pcd file;
  # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

```

x-Yabao, 2025-02-20:
关掉rviz了

Es1erda, 2025-02-20:
MARS_LVIG里基本都是large scale的数据，只有16g内存想维护一个large scale的global map是相当困难的。 可以参考郑博在另一个issue下的回复 #99 ：内存一直增长是因为维护的是global map，如果想让内存在一个范围内波动，可以设置 map_sliding_en: true 开启地图滑窗:。

xuankuzcr, 2025-02-20:
这个数据场景的尺度特别大，空间中的voxel数量会非常多，内存占用会很高，用笔记本或者台式机来跑吧。

RuifengLu, 2025-03-12:
> 配置文件如下：
>
> ```
> common:
>   img_topic: "/left_camera/image"
>   lid_topic: "/livox/lidar"
>   imu_topic: "/livox/imu"
>   img_en: 1
>   lidar_en: 1
>   ros_driver_bug_fix: false
>
> extrin_calib:
>   extrinsic_T: [...]
请问你解决这个问题了吗？