hku-mars/FAST-LIVO2 #103: How can I use gt_pose in simulator to replace _state?

Nangongyeee, 2025-02-20:
I tried to replace by using this code ,but it seems wrong while mapping,it looks like it didn't transfrom pointcloud from lidar to world:
`void LIVMapper::odom_callback(const nav_msgs::Odometry::ConstPtr& odom_msg) 
{
    std::lock_guard<std::mutex> lock(gt_pose_mutex);

    geometry_msgs::PoseStamped odom_pose_world;
    odom_pose_world.header = odom_msg->header;
    odom_pose_world.pose = odom_msg->pose.pose;

    geometry_msgs::PoseStamped odom_pose_camera_init;
    try 
    {
        tf_buffer.transform(odom_pose_world, odom_pose_camera_init, "camera_init");
    }
    catch (tf2::TransformException &ex) 
    {
        ROS_WARN("TF 变换失败: %s", ex.what());
        return;
    }
    latest_gt_pose.header = odom_pose_camera_init.header;
    latest_gt_pose.pose = odom_pose_camera_init.pose;

    tf::Quaternion q(
        odom_pose_camera_init.pose.orientation.x,
        odom_pose_camera_init.pose.orientation.y,
        odom_pose_camera_init.pose.orientation.z,
        odom_pose_camera_init.pose.orientation.w);
    tf::Matrix3x3 m(q);

    gt_state.rot_end << m[0][0], m[0][1], m[0][2],
                     m[1][0], m[1][1], m[1][2],
                     m[2][0], m[2][1], m[2][2];

    gt_state.pos_end << odom_msg->pose.pose.position.x,
                     odom_msg->pose.pose.position.y,
                     odom_msg->pose.pose.position.z;

    has_gt_pose = true;
    // ROS_DEBUG("Updated ground truth pose");
    std::cout << "Updated ground truth pose" << std::endl;
}`
`
void LIVMapper::processImu() 
{
  // double t0 = omp_get_wtime();
  gt_state_static = gt_state;
  _state = gt_state_static;
  p_imu->Process2(LidarMeasures, _state, feats_undistort);
  
  if (gravity_align_en) gravityAlignment();
  state_propagat = _state;
  voxelmap_manager->state_ = _state;
  voxelmap_manager->feats_undistort_ = feats_undistort;

  // double t_prop = omp_get_wtime();

  // std::cout << "[ Mapping ] feats_undistort: " << feats_undistort->size() << std::endl;
  // std::cout << "[ Mapping ] predict cov: " << _state.cov.diagonal().transpose() << std::endl;
  // std::cout << "[ Mapping ] predict sta: " << state_propagat.pos_end.transpose() << state_propagat.vel_end.transpose() << std::endl;
}
void LIVMapper::stateEstimationAndMapping() 
{
    if(!has_gt_pose) return;
  {
        std::lock_guard<std::mutex> lock(gt_pose_mutex);
        state_propagat = gt_state_static;
        _state = gt_state_static;
        voxelmap_manager->state_ = gt_state_static;
        // 状态已通过回调函数直接更新
  }
  switch (LidarMeasures.lio_vio_flg) 
  {
    case VIO:
      handleVIO();
      break;
    case LIO:
      handleLIO();
      break;
  }
}
void LIVMapper::handleLIO() 
{  
  _state = gt_state_static;  
  euler_cur = RotMtoEuler(_state.rot_end);
  fout_pre << setw(20) << LidarMeasures.last_lio_update_time - _first_lidar_time << " " << euler_cur.transpose() * 57.3 << " "
           << _state.pos_end.transpose() << " " << _state.vel_end.transpose() << " " << _state.bias_g.transpose() << " "
           << _state.bias_a.transpose() << " " << V3D(_state.inv_expo_time, 0, 0).transpose() << endl;
           
  if (feats_undistort->empty() || (feats_undistort == nullptr)) 
  {
    std::cout << "[ LIO ]: No point!!!" << std::endl;
    return;
  }
......
 voxelmap_manager->StateEstimation(state_propagat);

  // _state = voxelmap_manager->state_;

  _pv_list = voxelmap_manager->pv_list_;
......
}
`
`<node pkg="tf" type="static_transform_publisher" name="map2camera_init" args="0 0 0 0 0 3.14 /world /camera_init 100" />
`

[image]

[image]

xuankuzcr, 2025-02-20:
Hi, I’m not entirely clear about the issue you’re describing. Could you please clarify or provide more details about what’s not working as expected?

Nangongyeee, 2025-02-20:
非常感谢您的回复，我实际上是在做基于LIV的3DGS工作，这里我实际上只想使用fastlivo2提供的建图效果而不使用fastlivo2提供的位姿估计，于是想利用仿真中获得的真实位姿（gt_pose）来替代原本的位姿估计，我目前的方案就是完全用gt_pose来替换LIVMapper.cpp中的_state变量，但实际替换后发现lidar的点云似乎没有从lidar系变换到世界系，上面两幅图想表达的就是我的gtpose旋转的同时fastlivo2提供的点云也跟着一起旋转。应该是我的方法有问题，非常期待您的解答！

zhh2005757, 2025-02-20:
> 非常感谢您的回复，我实际上是在做基于LIV的3DGS工作，这里我实际上只想使用fastlivo2提供的建图效果而不使用fastlivo2提供的位姿估计，于是想利用仿真中获得的真实位姿（gt_pose）来替代原本的位姿估计，我目前的方案就是完全用gt_pose来替换LIVMapper.cpp中的_state变量，但实际替换后发现lidar的点云似乎没有从lidar系变换到世界系，上面两幅图想表达的就是我的gtpose旋转的同时fastlivo2提供的点云也跟着一起旋转。应该是我的方法有问题，非常期待您的解答！

我感觉你这种情况只需要用真值把激光雷达点云拼接起来，然后用图像附上颜色就行了，和算法没什么关系

YanhaoZhang, 2025-03-12:
您好，可以请问您的仿真环境是怎么搭建的吗？