hku-mars/FAST-LIVO2 #104: 关于使用mid360复现FASTLIVO2

shouleme521, 2025-02-22:
[image]
[image]
FASTLIVO2相比FASTLIVO在复现效果上有质的飞越，鲁棒性也强了好多，使用mid360在室内扫描后复现效果很好，重复扫描，墙壁几乎不会出现分层的现象，色彩也很鲜艳，最终成果可以直接用来三维建模。但是使用mid360在室外扫描一栋楼，不知道是是不是mid360点云较少，扫描时间短导致特征点少的原因，在扫描回到起点的时候发现有明显的分层现象，原因我会继续寻找。

fjz0911fang123, 2025-02-22:
把地图滑窗关了，就行了，不过内存吃的厉害

发自我的iPhone

   
0206.jpg (view on web)
 02065.jpg (view on web)

 
0206.jpg (view on web)
 02065.jpg (view on web)

shouleme521, 2025-02-22:
> 把地图滑窗关了，就行了，不过内存吃的厉害
> 
> 
> 
> 发自我的iPhone
> […](#)
> 
> ------------------ 原始邮件 ------------------
> 发件人: shouleme521 ***@***.***&gt;
> 发送时间: 2025年2月22日 20:49
> 收件人: hku-mars/FAST-LIVO2 ***@***.***&gt;
> 抄送: Subscribed ***@***.***&gt;
> 
> 
> 
> 
> 0206.jpg (view on web)
>  02065.jpg (view on web)
>  FASTLIVO2相比FASTLIVO在复现效果上有质的飞越，鲁棒性也强了好多，使用mid360在室内扫描后复现效果很好，重复扫描，墙壁几乎不会出现分层的现象，色彩也很鲜艳，最终成果可以直接用来三维建模。但是使用mid360在室外扫描一栋楼，不知道是是不是mid360点云较少，扫描时间短导致特征点少的原因，在扫描回到起点的时候发现有明显的分层现象，原因我会继续寻找。
> 
>   shouleme521 created an issue ([hku-mars/FAST-LIVO2#104](https://github.com/hku-mars/FAST-LIVO2/issues/104))
> 
> 0206.jpg (view on web)
>  02065.jpg (view on web)
>  FASTLIVO2相比FASTLIVO在复现效果上有质的飞越，鲁棒性也强了好多，使用mid360在室内扫描后复现效果很好，重复扫描，墙壁几乎不会出现分层的现象，色彩也很鲜艳，最终成果可以直接用来三维建模。但是使用mid360在室外扫描一栋楼，不知道是是不是mid360点云较少，扫描时间短导致特征点少的原因，在扫描回到起点的时候发现有明显的分层现象，原因我会继续寻找。

你问问他，地图滑窗是哪个选项，我的电脑是32G内存+4070

eung-ark, 2025-02-22:
How did you set up the camera? 
Looking at the image, it seems like you only turned around in place. 
I wonder which lens you used! 

Thank you.

shouleme521, 2025-02-22:
The radar was placed on the single-lens reflex camera tripod and rotated in place. The camera model is CU013-A0UC, and the lens is 6mm, but I don't know its model. @eung-ark

eung-ark, 2025-02-22:
If you don't mind, can I ask for a picture of MID360 setup equipment? 

I configured the setting with LIVOX AVIA first, and this time I'm going to configure MID360.

So I was just wondering how you fixed the MID360 and the camera at what angle!

Thank you very much.

shouleme521, 2025-02-22:
[image]
Tilt 27°
@eung-ark

20211105, 2025-02-23:
@shouleme521 请问你用的什么相机呀

mag72, 2025-02-28:
>Tilt 27°

@shouleme521 
Looks interesting. Like a fabric production, not 3D printed. Does the lid on the Mid-360 distort the point cloud?

Orangeflying, 2025-07-16:
> [image] Tilt 27° [@eung-ark](https://github.com/eung-ark)

Hey bro, can you share your structure file?