hku-mars/FAST-LIVO2 #107: 编译出错

lkrboom, 2025-02-25:
In file included from /home/Fast_LIVO2_ws/src/FAST-LIVO2/src/IMU_Processing.cpp:13:
/home/Fast_LIVO2_ws/src/FAST-LIVO2/include/IMU_Processing.h:52:12: error: field ‘fout_imu’ has incomplete type ‘std::ofstream’ {aka ‘std::basic_ofstream<char>’}
   52 |   ofstream fout_imu;
      |            ^~~~~~~~
In file included from /usr/include/c++/9/ios:38,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /usr/include/eigen3/Eigen/Core:96,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /home/Fast_LIVO2_ws/src/FAST-LIVO2/include/IMU_Processing.h:16,
                 from /home/Fast_LIVO2_ws/src/FAST-LIVO2/src/IMU_Processing.cpp:13:
/usr/include/c++/9/iosfwd:119:11: note: declaration of ‘std::ofstream’ {aka ‘class std::basic_ofstream<char>’}
  119 |     class basic_ofstream;
      |           ^~~~~~~~~~~~~~
make[2]: *** [FAST-LIVO2/CMakeFiles/imu_proc.dir/build.make:63: FAST-LIVO2/CMakeFiles/imu_proc.dir/src/IMU_Processing.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2147: FAST-LIVO2/CMakeFiles/imu_proc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/Fast_LIVO2_ws/src/FAST-LIVO2/include/LIVMapper.h:16,
                 from /home/Fast_LIVO2_ws/src/FAST-LIVO2/src/LIVMapper.cpp:13:
/home/Fast_LIVO2_ws/src/FAST-LIVO2/include/IMU_Processing.h:52:12: error: field ‘fout_imu’ has incomplete type ‘std::ofstream’ {aka ‘std::basic_ofstream<char>’}
   52 |   ofstream fout_imu;
      |            ^~~~~~~~
In file included from /usr/include/c++/9/ios:38,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /usr/include/eigen3/Eigen/Core:96,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /home/Fast_LIVO2_ws/src/FAST-LIVO2/include/IMU_Processing.h:16,
                 from /home/Fast_LIVO2_ws/src/FAST-LIVO2/include/LIVMapper.h:16,
                 from /home/Fast_LIVO2_ws/src/FAST-LIVO2/src/LIVMapper.cpp:13:
/usr/include/c++/9/iosfwd:119:11: note: declaration of ‘std::ofstream’ {aka ‘class std::basic_ofstream<char>’}
  119 |     class basic_ofstream;
这个是代码问题还是依赖问题呢?

Bob131649, 2025-02-25:
看起来像是c++版本问题

lkrboom, 2025-02-25:
include一个fstream的头文件就可以了

crazy-robotics, 2025-02-25:
I got the same issue while `catkin_make` I have following c++ version
```
orangepi@orangepi5plus:~$ g++ --version
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

crazy-robotics, 2025-02-25:
I solved the error by adding following header  
```
#include <fstream>
```
in 
```
/your_ws/FAST-LIVO2/include/IMU_Processing.h
``` 
and `catkin_make` again

crazy-robotics, 2025-02-25:
I got the following run time error #110  after successfully `catkin_make` it. @lkrboom do you got the same error?