hku-mars/FAST-LIVO2 #108: dataset gt

chongyu2021, 2025-02-25:
I noticed various sensor topics such as lidar (/livox/lidar), IMU (/livox/imu), and camera images (/left_camera/image), but I did not find any explicit ground truth data (e.g., trajectory or pose information) in the dataset. Could you confirm whether ground truth data is included, and if so, could you provide more details on which topics or files contain this information?

xuankuzcr, 2025-02-25:
The data does not contain ground truth. If needed, you can refer to our previous open-source dataset, [MARS-LVIG](https://mars.hku.hk/dataset.html).

chongyu2021, 2025-02-27:
thx

> The data does not contain ground truth. If needed, you can refer to our previous open-source dataset, [MARS-LVIG](https://mars.hku.hk/dataset.html).

closed 2025-02-27 by chongyu2021