hku-mars/FAST-LIVO2 #110: cmd /root/slam_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping

crazy-robotics, 2025-02-25:
I got following erro while launching `roslaunch fast_livo mapping_avia.launch`
```
root@orangepi5plus:~/slam_ws/src/FAST-LIVO2/launch# nano mapping_avia.launch  
root@orangepi5plus:~/slam_ws/src/FAST-LIVO2/launch# roslaunch fast_livo mapping_avia.launch
... logging to /root/.ros/log/d54df3ea-f369-11ef-9a9e-c0742bfdec54/roslaunch-orangepi5plus-2197.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://orangepi5plus:43445/

SUMMARY
========

PARAMETERS
 * /common/img_en: 0
 * /common/img_topic: /left_camera/image
 * /common/imu_topic: /livox/imu
 * /common/lid_topic: /livox/lidar
 * /common/lidar_en: 1
 * /common/ros_driver_bug_fix: False
 * /evo/pose_output_en: False
 * /evo/seq_name: CBD_Building_01
 * /extrin_calib/Pcl: [0.0194384, 0.104...
 * /extrin_calib/Rcl: [0.00610193, -0.9...
 * /extrin_calib/extrinsic_R: [1, 0, 0, 0, 1, 0...
 * /extrin_calib/extrinsic_T: [0.04165, 0.02326...
 * /imu/acc_cov: 0.5
 * /imu/b_acc_cov: 0.0001
 * /imu/b_gyr_cov: 0.0001
 * /imu/gyr_cov: 0.3
 * /imu/imu_en: True
 * /imu/imu_int_frame: 30
 * /laserMapping/cam_cx: 626.91359
 * /laserMapping/cam_cy: 522.799224
 * /laserMapping/cam_d0: -0.07616
 * /laserMapping/cam_d1: 0.123001
 * /laserMapping/cam_d2: -0.00113
 * /laserMapping/cam_d3: 0.000251
 * /laserMapping/cam_fx: 1293.56944
 * /laserMapping/cam_fy: 1293.3155
 * /laserMapping/cam_height: 1024
 * /laserMapping/cam_model: Pinhole
 * /laserMapping/cam_width: 1280
 * /laserMapping/scale: 0.5
 * /lio/beam_err: 0.05
 * /lio/dept_err: 0.02
 * /lio/layer_init_num: [5, 5, 5, 5, 5]
 * /lio/max_iterations: 5
 * /lio/max_layer: 2
 * /lio/max_points_num: 50
 * /lio/min_eigen_value: 0.0025
 * /lio/voxel_size: 0.5
 * /local_map/half_map_size: 100
 * /local_map/map_sliding_en: False
 * /local_map/sliding_thresh: 8
 * /pcd_save/colmap_output_en: False
 * /pcd_save/filter_size_pcd: 0.15
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: False
 * /preprocess/blind: 0.8
 * /preprocess/filter_size_surf: 0.1
 * /preprocess/lidar_type: 1
 * /preprocess/point_filter_num: 1
 * /preprocess/scan_line: 6
 * /publish/blind_rgb_points: 0.0
 * /publish/dense_map_en: True
 * /publish/pub_effect_point_en: False
 * /publish/pub_plane_en: False
 * /publish/pub_scan_num: 1
 * /rosdistro: noetic
 * /rosversion: 1.17.0
 * /time_offset/exposure_time_init: 0.0
 * /time_offset/img_time_offset: 0.1
 * /time_offset/imu_time_offset: 0.0
 * /uav/gravity_align_en: False
 * /uav/imu_rate_odom: False
 * /vio/exposure_estimate_en: True
 * /vio/img_point_cov: 100
 * /vio/inv_expo_cov: 0.1
 * /vio/inverse_composition_en: False
 * /vio/max_iterations: 5
 * /vio/normal_en: True
 * /vio/outlier_threshold: 1000
 * /vio/patch_pyrimid_level: 4
 * /vio/patch_size: 8
 * /vio/raycast_en: False

NODES
  /
    laserMapping (fast_livo/fastlivo_mapping)

ROS_MASTER_URI=http://localhost:11311

process[laserMapping-1]: started with pid [2212]
[ INFO] [1740482850.074075605]: Found parameter: laserMapping/cam_model, value: Pinhole
[ INFO] [1740482850.076991653]: Found parameter: laserMapping/cam_width, value: 1280
[ INFO] [1740482850.077368482]: Found parameter: laserMapping/cam_height, value: 1024
[ INFO] [1740482850.077817352]: Found parameter: laserMapping/scale, value: 0.5
[ INFO] [1740482850.078236472]: Found parameter: laserMapping/cam_fx, value: 1293.57
[ INFO] [1740482850.078901756]: Found parameter: laserMapping/cam_fy, value: 1293.32
[ INFO] [1740482850.079279460]: Found parameter: laserMapping/cam_cx, value: 626.914
[ INFO] [1740482850.079695955]: Found parameter: laserMapping/cam_cy, value: 522.799
[ INFO] [1740482850.080096991]: Found parameter: laserMapping/cam_d0, value: -0.07616
[ INFO] [1740482850.080488403]: Found parameter: laserMapping/cam_d1, value: 0.123001
[ INFO] [1740482850.080865816]: Found parameter: laserMapping/cam_d2, value: -0.00113
[ INFO] [1740482850.081232145]: Found parameter: laserMapping/cam_d3, value: 0.000251
scale: 0.5
[laserMapping-1] process has died [pid 2212, exit code -11, cmd /root/slam_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/root/.ros/log/d54df3ea-f369-11ef-9a9e-c0742bfdec54/laserMapping-1.log].
log file: /root/.ros/log/d54df3ea-f369-11ef-9a9e-c0742bfdec54/laserMapping-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

xuankuzcr, 2025-02-25:
As mentioned in hku-mars/FAST-LIVO#53, try using GDB to locate the error.

lkrboom, 2025-02-26:
opencv too new,perhaps.

crazy-robotics, 2025-02-26:
> As mentioned in [hku-mars/FAST-LIVO#53](https://github.com/hku-mars/FAST-LIVO/issues/53), try using GDB to locate the error.

I tried it but the problem is same but with more extra line :)

crazy-robotics, 2025-02-26:
> opencv too new,perhaps.

Same version of opencv is working in my pc but not in `Jetson orin nano` and `Orange Pi 5 pluse` but  `fast-livo2` compiled successfully

tanble, 2025-02-26:
On the Jetson Orin platform, please switch the OpenCV version to 4.2. I have tried versions 3.4 and 4.5 before, but neither worked.