hku-mars/FAST-LIVO2 #112: 雷达lio观测模块增量更新不理解

DeionL, 2025-02-26:
solution = K_1.block<DIM_STATE, 6> (0, 0) * HTz + vec.block<DIM_STATE, 1> (0, 0) - G.block<DIM_STATE, 6> (0, 0) * vec.block<6, 1> (0, 0);

为什么后面加上vec.block<DIM_STATE, 1> (0, 0) - G.block<DIM_STATE, 6> (0, 0) * vec.block<6, 1> (0, 0);不是很理解，麻烦解答一下

mattcenoo, 2025-03-01:
[image]

可以看下fast-lio论文，lio部分是一样的，省略了Jk

biggiantpigeon, 2025-05-28:
> <img alt="Image" width="777" src="https://private-user-images.githubusercontent.com/26203247/418236594-03ee4313-297b-49f7-948b-824ca528a117.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg0MjkxNTksIm5iZiI6MTc0ODQyODg1OSwicGF0aCI6Ii8yNjIwMzI0Ny80MTgyMzY1OTQtMDNlZTQzMTMtMjk3Yi00OWY3LTk0OGItODI0Y2E1MjhhMTE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTI4VDEwNDA1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU4MTEzNDhlYzVlOGUwNGZiYmFlMTQxNzMwODg1OGQyZjM2NWQ2OTQ3YTRlYmVhMTU2YmY5NzZkMmQ3NTBmN2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.w7wlR0VZXp8zxSPjuDPcHneMoo7WtGgduLDPM1qP4B8">
> 可以看下fast-lio论文，lio部分是一样的，省略了Jk

我不理解为什么xk+1-Xk的更新，对应的是-Kz？总感觉这个公式右边括号里多了个负号。公式跟代码倒是符合的，因为用的vec是反的。