hku-mars/FAST-LIVO2 #114: 请教关于vio更新的雅可比推导

mattcenoo, 2025-02-27:
郑博你好，感谢你的工作，我在vio更新updateStateInverse下的推导遇到了些问题，请帮忙给个思路
为什么precomputeReferencePatches中对参考patch的雅可比是这样的

[image]

然后求观测对[image]
的雅可比时，为什么是这样的

`

          MD(1, 3) J_dR = H_sub_inv.block<1, 3>(i * patch_size_total + x * patch_size + y, 0);
          
          MD(1, 3) J_dt = H_sub_inv.block<1, 3>(i * patch_size_total + x * patch_size + y, 3);

          JdR = J_dR * Rwi + J_dt * P_wi_hat * Rwi;

          Jdt = J_dt * Rwi;
`
试了各种推导都得不到这个结果，请郑博指导

xuankuzcr, 2025-03-01:
感谢支持，可以参考下这个issue [#80](https://github.com/hku-mars/FAST-LIVO/issues/80)

mattcenoo, 2025-03-03:
[image]

能凑上JdR, 但Jdt的凑不上，而且Rcw的c是参考帧，Rwi又是当前帧，有点懵

[image]

delta T是放在ui prime上的，yc对他求导是不是应该有个负号
然后感觉是相对于参考帧的delta T, 为什么solution可以直接加到state上
还得请郑博指导一下，谢谢 @xuankuzcr

ZJW-HUST, 2025-08-11:
从代码上看好像是这样：$$ -R_{cw}R_{wi} = -R_{ci}R_{iw}^{ref}R_{wi}^{cur}=-R_{ci}R_{i_{ref}i_{cur}} $$，主要不清楚具体的残差函数是啥，看论文是说是使用的ref patch部分的位姿增量