hku-mars/FAST-LIVO2 #116: How to use the OS1-16 gen 1 from Ouster in fastlivo2

JackCooper666, 2025-02-27:
Hello, I read your excellent work, and get a clear implementation from your demo dataset. Now, I am trying to use the fastlivo2 in the NTU-VIRAL dataset, but their lidar is OS1-16 gen from Ouster, so I adjust the avia.yaml like following. Then I roslaunch the fastlivo2 and play the eee01.bag by rosbag play, but nothing happen, the rviz is empty while the launch file did not show me any error.  Thanks for your help.

[image]

[image]

common:
  img_topic: "/left_camera/image"
  lid_topic: "/os1_cloud_node1/points" 
  imu_topic: "/imu/imu" 
  img_en: 1
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [0.05, 0, -0.055]
  extrinsic_R: [1, 0, 0, 
  0, 1, 0, 
  0, 0, 1]
  
 
  Rcl: [0.02183, 0.9997, -0.002,
            -0.01312,  0.0023, 0.9999,
            0.9997,  -0.022,  0.0132] 
  Pcl: [0.01, 0.04, -0.105] 

time_offset: 
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess:
  point_filter_num: 1
  filter_size_surf: 0.1
  lidar_type: 3 
  scan_line: 16
  blind: 2

vio:
  max_iterations: 5
  outlier_threshold: 1000 
  img_point_cov: 100
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: true
  imu_int_frame: 30
  acc_cov: 0.5 # 0.2
  gyr_cov: 0.3 # 0.5
  b_acc_cov: 0.0001 
  b_gyr_cov: 0.0001 

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.05
  min_eigen_value: 0.0025 
  voxel_size: 0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:  
  map_sliding_en: false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: false

publish:
  dense_map_en: true
  pub_effect_point_en: false 
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  seq_name: "CBD_Building_01"
  pose_output_en: false

pcd_save:
  pcd_save_en: false
  colmap_output_en: false 
  filter_size_pcd: 0.15
  interval: -1

closed 2025-02-28 by JackCooper666

xuankuzcr, 2025-03-01:
Hi, it seems you entered the wrong camera topic name.

xuankuzcr, 2025-03-29:
Issue #192 has resolved your problem @JackCooper666